SpinalHDL/SpinalHDL #806: feat: Blackbox generator

numero-744, 2022-07-25:
Hello,

The company I am currently working in finds SpinalHDL interesting for their top-level, to assemble all the components, because of the bus abstraction which generates normal wires in RTL. (If someone tells you that you should generate records instead, please don't!)

I think a feature that could be really interesting for that is to generate blackboxes from VHDL / SystemVerilog. It could be a function to put in a `main`, or a `main` itself (or both?), delivered with SpinalHDL.

I could work on this topic on my personal time, but I would first know what you think about it.

It seems work has been already done on this topic, but not merged: https://github.com/SpinalHDL/SpinalHDL/pull/318 and https://github.com/SpinalHDL/SpinalHDL/pull/789

# MVP features

- [ ] Take a file `$f` as an argument, say `$filename.$extensions`
- [ ] Detect language of `$f` based on `$extension` (`v` or `sv`: parse as SystemVerilog, `vhd`: parse as VHDL`
- [ ] Parse `$f` with the correct parser, implemented for instance using [`Combinators`](https://github.com/scala/scala-parser-combinators)
- [ ] List all modules/entities detected
- [ ] For each one, list all its ports
- [ ] Write into `src/main/scala/blackboxes/$filename.scala` the code defining a blackbox for each module/entity found during above steps

# Other features

- [ ] Take an optional list of SpinalHDL buses as a second argument, and find in the signals the ones with names, sizes and directions matching with the signals of the bus. Print warnings/errors if some are incorrect
- [ ] Work with a specific module/entity in a file
- [ ] Choose the output filename

# Potential issues

- It would not manage complex cases with `` `include ``s
- It would only manage simple connections
- Other things?

kleinai, 2022-07-31:
Hi,
There's definitely a number of users who want these features. I'm also interested myself.

I added record support for VHDL a while back to help with blackboxing complex IP. Also wouldn't recommend it for use in Spinal but it would be nice to parse records and structs in the future.

Maybe we could move some of these Spinal based tools to a different package or repo?

numero-744, 2022-08-01:
> I added record support for VHDL a while back to help with blackboxing complex IP

It does not seem to be documented in RTD, would you have a minimal example to provide, please? (VHDL/SV declaration + the SpinalHDL blackbox code which should be generated for it)

An issue I see is that types are often defined in a different file (a `*_pkg.vhd` file in VHDL for instance) and the blackbox generator would not be able to know which files are compiled and in which order.

> Maybe we could move some of these Spinal based tools to a different package or repo?

Sure, it could be too heavy to be released in SpinalHDL, but I really would like to be able to generate SpinalHDL buses, including custom ones. So it must at least be written in Scala and will depend on SpinalHDL.

kleinai, 2022-08-02:
I forgot to document it at the time. It worked something along the lines of:

```scala
class MyStruct extends SpinalStruct("struct_type") {
  val a = UInt(4 bit)
  val b = UInt(2 bit)
}

class MyBB extends BlackBox {
  val io = new Bundle {
    val struct_data = in(new MyStruct)
    val c = out UInt(6 bit)
  }
}

class MyTop extends Component {
  val io = new Bundle {
    val a = in UInt(4 bit)
    val b = in UInt(2 bit)
    val c = out UInt(6 bit)
  }

  val bb = new MyBB()
  io.a <> bb.io.struct_data.a
  io.b <> bb.io.struct_data.b
  io.c <> bb.io.c
}

object MyTop {
  def main(args: Array[String]): Unit = {
    SpinalVhdl(new MyTop)
  }
}
```

Results in:
```vhdl
.. snip ..
entity MyTop is
  port(
    io_a : in unsigned(3 downto 0);
    io_b : in unsigned(1 downto 0);
    io_c : out unsigned(5 downto 0)
  );
end MyTop;

architecture arch of MyTop is
  signal bb_io_struct_data : struct_type;
  signal bb_io_c : unsigned(5 downto 0);

  component MyBB is
    port( 
      io_struct_data : in struct_type;
      io_c : out unsigned 
    );
  end component;
  

begin
  bb : MyBB
    port map ( 
      io_struct_data.a => io_a,
      io_struct_data.b => io_b,
      io_c => bb_io_c 
    );
  io_c <= bb_io_c;
end arch;
```

The actual record type would need to be provided by the black box sources.

It looks like there's still a bug. `bb_io_struct_data` should be suffixed not the names in the instance port. I think that's not supported in older VHDL versions.

Full disclosure, I never ended up using `SpinalStruct` in any real code.