SpinalHDL/SpinalHDL #811: Verilator Caseoverlap from generated Code

saahm, 2022-08-02:
Hey,

the following code causes warnings in Verilator 4.107 devel rev v4.106-62-g82d8fe0c and SpinalHDL 1.7 (also older) but not on SpinalHDL 1.4.

```scala
package mylib.duplicatecase

import spinal.core._
import spinal.lib._

import spinal.sim._
import spinal.core.sim._

object MiaoCode {
    def F4_FOO    = M"0000"
    def F4_BAR    = M"0001"
    def F4_MIAU   = M"0010"
    def F4_MOO    = M"0011"
    def F4_WOOF   = M"0100"
}
object MooCode {
    def F4_FOOBAR = M"0000"
    def F4_BARFOO = M"0001"
}

object FooEnumType extends SpinalEnum{
  val undef, fooFoo, fooBar, fooMiau, barFoo, barBar, barMiau = newElement()
}

//Hardware definition
class Bar extends Component {
    val io = new Bundle {
        var myIn = in Bits(5 bits)
        var myFoo = out(FooEnumType())
        val myOut1 = out Bool()
        val barFoo = out Bool()
    }

    val op = Bits(4 bits)
    op := io.myIn(3 downto 0)

    io.myOut1 := False
    io.barFoo := False
    import MooCode._, MiaoCode._
    switch(op){
        is(F4_FOO, F4_FOOBAR){
            io.myFoo := FooEnumType.fooBar
            io.myOut1 := True
        }
        is(F4_MIAU){
            io.myFoo := FooEnumType.fooMiau
        }
        default{
            io.myFoo := FooEnumType.barMiau
            io.barFoo := True
        }
    }

}

object MyBarSim {
  def main(args: Array[String]) {
    SimConfig.withWave.doSim(new Bar){dut =>
      //Fork a process to generate the reset and the clock on the dut
      // dut.clockDomain.forkStimulus(period = 10)
      
      //Wait a rising edge on the clock
      // dut.clockDomain.waitRisingEdge()
      sleep(1)

    }
  }
}
object MyBarVerilog {
  def main(args: Array[String]) {
    SpinalConfig(
      targetDirectory = "rtl/"
      ).generateVerilog(new Bar)
      .printPruned()
  }
}
```

When generating the Verilog (MyBarVerilog) there is no issue. Calling the simulation (MyBarSim) causes Verialtor to have `%Warning-CASEOVERLAP`.

If the constants inside MiaoCode and MooCode are B instead of M SpinalHDL can generate an error in generation time already.

Overlapping cases shouldnt really cause any issue as its just "duplicate code". Maybe this should cause a warning? The warning can be ommited with `/* verilator lint_off CASEOVERLAP*/` and `/* verilator lint_on CASEOVERLAP*/` around such overlapping cases. Nonetheless it might be possible to generate unreachable or overlapping cases if they are not inside the same "is(...)" statement within SpinalHDL (not tested).

I initially got this in a CPU design where i have an extra decoding check for a different instruction using the same opcode, causing the is(...) case to have the same constants being checked. I'd like to keep that design that way as its quite a big one and the readability could suffer if I have to omit it by hand or do something else.

Dolu1990, 2022-08-03:
Thanks ^^

Got both  B"0000" and M"0000" from your example catched with : 

DUPLICATED ELEMENTS IN SWITCH IS(...) STATEMENT. value=0

There is cases where catching duplication is a good thing.

now you can do the following to turn of the error :

```scala
    switch(op, strict = false){
``` 

This will remove the duplications

closed 2022-08-03 by Dolu1990

saahm, 2022-08-03:
Will it cause an Error still or Warning?

commit a130f7b7 (2022-08-03): fix #811 switch { is(0, 0) } are now properly detected