SpinalHDL/SpinalHDL #813: Logic to generating pastValid would cause error when prove.

Readon, 2022-08-05:
The generated code for pastValid in SpinalHDL now is
``` verilog
always @(posedge clk or posedge reset) begin
   if(reset) begin
     formal_with_past <= 1'b0
   end else begin
     formal_with_past <= (!reset)
   end
```
The logic works good when doing BMC check. 
However, when it is in prove mode, the status of formal_with_past could be any value do not obey the asserts and assumes.
When the formal_with_past is initially set to 0 then prove logic would fail.

According to the code that ZipCpu [opened](https://github.com/ZipCPU/wb2axip/blob/ea0fe7fc386353451515a7d760df525c7d617b06/bench/formal/faxil_master.v#L172), use initial keyword to set the formal_with_past would solve the problem.

Dolu1990, 2022-08-08:
Ahhh, so we could adding a xxx.initial(False)  would fix it right ?
Could you give a try ?

Readon, 2022-08-08:
> 

I think it is related with the generating method of pastValid.
Adding xxx.initial(False) is also a more generic solution.

I will write a minimal example to test it first.

Readon, 2022-08-16:
I have test with [rtcclock/rtcstopwatch](https://github.com/ZipCPU/rtcclock/blob/0bc3c7fe83fbb17d7e2fbc0dbf8166e5ab2b6734/bench/formal/rtcstopwatch.sby#L1)
by replace the f_past_valid logic with above then validation failed.

The correct version should be
``` verilog
	reg	f_past_valid;
	initial	f_past_valid = 1'b0;
	always @(posedge i_clk)
	   f_past_valid <= 1'b1;
```

So there would be two targets.
1. modify generating mechanism of pastValid to generate correct formal_with_past signal.
2. add initial method for datatypes to fit every signal requires specific initial status.

I am not familiar with those topic, do you might add these? @Dolu1990

Dolu1990, 2022-08-17:
Shouldn't we also include that on reset f_past_valid should be cleared ? in addition to the initial clear, to be very exaustive and not too specific ?

So : 
```scala
  def pastValid() : Bool = signalCache("formal.pastValid")(past(!ClockDomain.current.isResetActive) init(False) initial(False) setWeakName("formal_with_past"))
``` 

The "initial" feature is already implemented :)

Doesn't it fix all the issues ?

Readon, 2022-08-17:
> ```scala
>  initial(False)
> ```

The problem is that the reg would have value False when prove started, and reset is deasserted. 

Unless we assert that
``` scala
    when(!ClockDomain.current.isResetActive) assert(pastValid)
```
However, this assertion is not true while running BMC.

Dolu1990, 2022-08-17:
So 

```verilog
	reg	f_past_valid;
	initial	f_past_valid = 1'b0;
	always @(posedge i_clk)
	   f_past_valid <= 1'b1;
```

=> 

```scala
RegNext(True) initial(False)  setWeakName("f_past_valid")	   
``` 

=>  (Generated verilog)

```verilog
  reg                 formal_with_past;

  initial begin
    formal_with_past = 1'b0;
  end

  always @(posedge clk) begin
    formal_with_past <= 1'b1;
  end
```

I'm realy not knowledgable in formal stuff, It seems for me that depending the formal verification mode, we may want different implementation of "pastValid"

Let's me know what you think is the best solution, i kind of have not much vision.

Readon, 2022-08-17:
> So
>
> ```verilog
> 	reg	f_past_valid;
> 	initial	f_past_valid = 1'b0;
> 	always @(posedge i_clk)
> 	   f_past_valid <= 1'b1;
> ```
>
> =>
>
> ```scala
> RegNext(True) initial(False)  setWeakName("f_past_valid") [...]

I have reviewed the code ZipCPU provided and related trace files.
The f_past_valid is a signal that demonstrated past value is existed, there should be one valid value to be checked.

By definition, it has **no relation with reset** at all, there should be extra code to work with that.
So 
``` scala
    RegNext(True) initial(False)  setWeakName("f_past_valid")
```
is alright, if we take their implementation as reference.
If then, the existing implementation in spinalHDL should be named as pastValidAfterReset.

Dolu1990, 2022-08-19:
> By definition, it has no relation with reset at all, there should be extra code to work with that.

Does it make sense to trigger on pastValid when the system is under reset ? I mean, in terms of usefullness. To me, system under reset => void. I can see that for instance you would like to assert that a given signal is low under reset, as for instance the valid of a stream, in which case it can make sense.

But i guess that the vast majority of the time, you want to assert stuff outside of reset right ? so maybe pastValid should be with reset, and we could have a pastValidIncludingReset or something like that ?

> if we take their implementation as reference.

We do not necessarly have to, not saying we should'nt

Readon, 2022-08-20:
> Does it make sense to trigger on pastValid when the system is under reset ? I mean, in terms of usefullness. To me, system under reset => void. I can see that for instance you would like to assert [...]

The existing pastValid is just a synchronous version of "!reset", so it's not necessarily to define a new one.

Do you know the original intention of defining pastValid?
Literally, pastValid should just mean that past values ​​are valid.

Dolu1990, 2022-08-22:
> Do you know the original intention of defining pastValid?
Literally, pastValid should just mean that past values ​​are valid.

Yes, i guess i have some bias because i see it from a syncronous manner ^^

So, i guess pastValid + pastValidAfterReset is fair

Dolu1990, 2022-08-22:
merged ^^

closed 2022-08-22 by Dolu1990

Readon, 2022-09-03:
One more thing, would be notice here.
When verify an async logic, the clocks are generated from global clock where pastValid would be designwise valid after the first global clock. 
The current pastValid is reasonably good.
However, pastValidAfterReset might lead error because it related with reset while global clock has no reset, cause the resetKind = BOOT.

Dolu1990, 2022-09-05:
Wouldn't the pastValidAfterReset also use the resetKind = BOOT ? => initial reg = 0

Readon, 2022-09-05:
> Wouldn't the pastValidAfterReset also use the resetKind = BOOT ? => initial reg = 0

The thing is that there are no reset for global clock domain.
But pastValidAfterReset is related with reset, there could be more than one resets in the design, which one to use is a problem. 
As I understand it should be derived from the reset in current clock domain so it's clock domain related.
However, in current version it use "clockDomain.current" clock as default.

Dolu1990, 2022-09-05:
Ahhh damned, the signal cache should also use the clock domain to create the key. that's one issue. i'm fixing it now.

Dolu1990, 2022-09-05:
Fixed in https://github.com/SpinalHDL/SpinalHDL/commit/2c20f1c8d9a29d595e618ac30fada3812feb5fea

Readon, 2022-09-05:
> Fixed in [2c20f1c](https://github.com/SpinalHDL/SpinalHDL/commit/2c20f1c8d9a29d595e618ac30fada3812feb5fea)

It's a little bit complicated, the name of those signal should be different with clock domain too.
And the pastValid signal could be design wise, because in cross clock design, all clocks and resets are driven by the global clock,
So if there is one global clock rising, then the pastValid is valid because there is already past values there.

Dolu1990, 2022-09-05:
> It's a little bit complicated, the name of those signal should be different with clock domain too.

So, as the name is set with setWeakName, name conflicts should be renamed with _1 _2 and so on. I guess it isn't very important to have individual precise name here. (lazyness :D)

> So if there is one global clock rising, then the pastValid is valid because there is already past values there.

Yes that's right. would have to add as key the clock signal, clock enable, clock polarity and clock enable polarity

Readon, 2022-09-05:
> > It's a little bit complicated, the name of those signal should be different with clock domain too.
> 
> So, as the name is set with setWeakName, name conflicts should be renamed with _1 _2 and so on. I guess it isn't very important to have individual precise name here. (lazyness :D)

I just pulled and tried, it works.

> 
> > So if there is one global clock rising, then the pastValid is valid because there is already past values there.
> 
> Yes that's right. would have to add as key the clock signal, clock enable, clock polarity and clock enable polarity

Emmm, the global clock is always enabled, it's not necessarily to check that.
But in synchronous version it could be.