SpinalHDL/SpinalHDL #816: `Exception in thread "main" java.lang.Exception: ISim engine error: Failed to Load up XSI.`

dzx-dzx, 2022-08-08:
```
object ... {
  def main(args: Array[String]): Unit = {
    ...
    val compiled = SimConfig.withWave
      .withXSim
      .withXSimSourcesPaths(ArrayBuffer("..."),ArrayBuffer(""))
      .compile(new ...)
      .doSim(dut => {
        simSuccess()
      })
  }
}
```
This has led to:
```
...
[info] [Progress] XSIM compilation started
[error] Exception in thread "main" java.lang.Exception: ISim engine error: Failed to Load up XSI.
[error]         at spinal.sim.xsi.JNIXSIIfaceJNI.new_XSIIface(Native Method)
[error]         at spinal.sim.xsi.XSIIface.<init>(XSIIface.java:40)
[error]         at spinal.sim.XSimBackend.getInterface(XSimBackend.scala:312)
[error]         at spinal.sim.SimXSim.<init>(SimXSim.scala:9)
[error]         at spinal.core.sim.SpinalSimConfig$$anon$8.newSimRaw(SimBootstraps.scala:1005)
[error]         at spinal.core.sim.SimCompiled.doSimApi(SimBootstraps.scala:521)
[error]         at spinal.core.sim.SimCompiled.doSim(SimBootstraps.scala:498)
...
```

kleinai, 2022-08-08:
What version of SpinalHDL and Vivado are you using?

dzx-dzx, 2022-08-09:
@kleinai SpinalHDL `1.7.2`, VIVADO `2019.1`.

dzx-dzx, 2022-08-09:
There would be a similar, but different exception when simulating with `2022.1`:
```
[info] [Progress] XSIM compilation started
[info] Design lib: xsim.dir/udp_40G_TOP_behav/xsimk.so
[info] Kernel lib: librdi_simulator_kernel.so
[error] Exception in thread "main" java.lang.Exception: basic_string::_M_construct null not valid
[error]         at spinal.sim.xsi.JNIXSIIfaceJNI.new_XSIIface(Native Method)
[error]         at spinal.sim.xsi.XSIIface.<init>(XSIIface.java:40)
[error]         at spinal.sim.XSimBackend.getInterface(XSimBackend.scala:312)
[error]         at spinal.sim.SimXSim.<init>(SimXSim.scala:9)
[error]         at spinal.core.sim.SpinalSimConfig$$anon$8.newSimRaw(SimBootstraps.scala:1005)
[error]         at spinal.core.sim.SimCompiled.doSimApi(SimBootstraps.scala:521)
[error]         at spinal.core.sim.SimCompiled.doSim(SimBootstraps.scala:498)
```

kleinai, 2022-08-09:
Your first error failed to load the XSI all together. I'm guessing that older version has some differences in XSI interface functions.

Your second error makes me think something isn't configured right. Could you look for a `spinal_xsim.h` in your sim project?

dzx-dzx, 2022-08-10:
Here's the content:
```
#define SIM_KERNEL "librdi_simulator_kernel.so"
#define SIM_DESIGN "xsim.dir/udp_40G_TOP_behav/xsimk.so"
#define SIM_WAVE   "test.wdb"
#define SIM_HAS_WAVE true

```

kleinai, 2022-08-10:
Okay, nothing out of the ordinary there. I'm installing 2022.1 and will follow up.

kleinai, 2022-08-11:
Alright, I've tried with both Windows and Linux 2022.1 and have no problems simulating.

Have you tried simulating code without any Vivado IP?

dzx-dzx, 2022-08-11:
That would be successful, yes.
May I share my source code?

kleinai, 2022-08-11:
I'm on the free Vivado version so I likely cannot help you. I'd suggest narrowing it down to an IP that causes the error. Maybe it's a generic that's set wrong?

dzx-dzx, 2022-08-11:
I see...Thanks.

dzx-dzx, 2022-08-18:
I think I found the problem. It's the `dumpwave` function which inserts a snippet to the Verilog source. This helps to export waveform when simulating with iverilog, but it stalls Vivado.
Not related to SpinalHDL itself, but might be a headsup.

kleinai, 2022-08-18:
Interesting! A Vivado bug. I don't typically run with `.dumpWave` so that's probably why I didn't catch it. Do you mind sharing what option you think broke it?

dzx-dzx, 2022-08-19:
Hmm...I'm sorry don't follow you. `dumpwave` is the option that is causing the problem. To be specific, with this the following will be inserted to the code:
```
initial begin
    $dumpfile("wave.vcd");
    $dumpvars(0, ***);
  end
```
Oh, I didn't provide any parameter to it, if that's what you mean.

numero-744, 2022-11-15:
So can this issue be closed?

dzx-dzx, 2022-11-16:
> So can this issue be closed?

Oh yes, sure. But is using `dumpWave` crashes the `xsim` an expected behavior?

numero-744, 2022-11-16:
Ahh I though it was fixed XD
Well if `xsim` does not implement it, I don't know if we can consider it a Spinal bug… Maybe an issue should be opened on their side?

@kleinai what do you think about it?

kleinai, 2022-11-16:
> > So can this issue be closed?
> 
> Oh yes, sure. But is using `dumpWave` crashes the `xsim` an expected behavior?

No, `xsim` crashing is not the expected behavior.

I booted one of the standard Spinal tests with XSim, dumpWave in the SpinalConfig, the `.withWave` option. The `wave.vcd` generated without issue and looks valid in GTKwave.

Your problem may be with `xsim` itself or an IP.  `simWorkspace` should have a subdirectory for your simulation. In there will be vivado sim generation logs. There are runtime logs in `xsim.dir` you could look for as well.

dzx-dzx, 2022-11-19:
> Your problem may be with `xsim` itself or an IP.

Exactly the latter. I can't remember much but I don't think I had discovered anything useful when I was looking into the issue... I will try next time. Thanks.

closed 2022-11-21 by dzx-dzx