SpinalHDL/SpinalHDL #817: Possible delta-cycle issue with SpinalSim and verilator

andreasWallner, 2022-08-08:
When simulating the HDL shown below I would expect the `rxOld` signal to be a delayed version of the `io.read` (I actually stumbled upon this using `Bool.edge`), but looking at the waveforms in gtkwave it seems that there might be an issue with how verilator and SpinalSim interact. For the first edge this works as expected, but further edge don't. `io.read` is generated in the testbench inside a `fork`ed section.

Looking at some debug output I added it seems that the difference between the working and not working edge is the order in which delta cycles are processed: On the first (working edge) the signal assignment is done after the clock edge, in the second (not working edge) the signal assignment happens before the clock edge is processed.

Is this just a misuse of `fork` or an issue?

[image]

When simulating:
``` scala
import spinal.sim._
import spinal.core._
import spinal.core.sim._
import spinal.lib._
import spinal.lib.sim._
import org.scalatest.funsuite.AnyFunSuite

case class Master() extends Component {
  val io = new Bundle {
    val write = out Bool() setAsReg
    val read = in Bool()

    val trigger = new Bundle {
      val rx = in Bool()
    }

  }
  io.write init True
  io.write := !io.trigger.rx

  val readOld = RegNext(io.read)
  val edge = io.read ^ readOld
  readOld simPublic()
  edge simPublic()
}

class CoreSim extends AnyFunSuite {
  val dut = SimConfig.withWave.compile(Master())

  val etu = 20

  test("rx byte") {
    dut.doSim("rx byte") { dut =>
      SimTimeout(3000)

      dut.io.read #= true
      dut.io.trigger.rx #= false

      dut.clockDomain.forkStimulus(10)
      dut.clockDomain.waitSampling(3)

      forkSensitive {
        println(f"${simTime()} ${dut.io.read.toBoolean} ${dut.readOld.toBoolean} ${dut.edge.toBoolean}")
      }
      dut.clockDomain.onSamplings {
        println(f"clock ${simTime()}")
      }

      fork {
        waitUntil(dut.io.write.toBoolean == false)
        for(_ <- 0 to 4) {
          dut.io.read #= !dut.io.read.toBoolean
          sleep(etu)
        }
      }
      dut.io.trigger.rx #= true
      dut.clockDomain.waitActiveEdge()
      dut.io.trigger.rx #= false

      sleep(200)
    }
  }
}
```

Debug output snip:
```
...
195 true true false
200 true true false
clock 200
200 true true false
200 false true false
200 false true true
205 false true true
...
235 true true false
240 false true false
clock 240
240 false false false
245 false false false
```

Dolu1990, 2022-08-10:
To me it seems normal.

the code will pass waitUntil(dut.io.write.toBoolean == false) after the rising edge, but subsequant sleep(etu) will pass at the beginning of the scheduled timestamp (first delta cycle), and so you will assign dut.io.read at the same time than the clock is assigned high, so flip flop will see both assignements happening at the same time

andreasWallner, 2022-08-13:
Thinking about it for a second time it makes sense now - when I wrote the bug I somehow mixed up the delta cycles...

closed 2022-08-13 by andreasWallner

Dolu1990, 2022-08-15:
No worries ^^
I would say in general, to never do asyncronous wait in the testbench (waitUntil), but instead do things as cd.onSamplings :)