SpinalHDL/SpinalHDL #820: Area Hierarchy Bug

dokleina, 2022-08-12:
I'm seeing what I believe is a bug in hierarchy check. I have an Area that (as far as I understand) isn't doing anything invalid.

I put together this demo:
```scala
object HierarchyBug {

  class TestClass[T <: Data](streamPayload : HardType[T]) extends Area {
    val io = new Bundle {
      val inputStream = Stream(streamPayload)
      val outBits     = Bits(streamPayload.getBitsWidth bits)
    }

    val flow = io.inputStream.stage().toFlow

    io.outBits := flow.payload.as(Bits(8 bits))
  }

  class TopLevel extends Component {
    val io = new Bundle {
      val stream = slave Stream(Bits(8 bits))
      val output = out Bits(8 bits)
    }
    val testArea = new TestClass(io.stream.payload)
    testArea.io.inputStream << io.stream
    io.output := RegNext(testArea.io.outBits)
  }

  def main(args: Array[String]): Unit = {
    SpinalVhdl(new TopLevel())
  }
}
```

And the error I get is:
```
Exception in thread "main" spinal.core.SpinalExit: 

 Error detected in phase PhaseCheckHiearchy
********************************************************************************
********************************************************************************

HIERARCHY VIOLATION : (toplevel/??? : in Bool) is driven by (toplevel/when_Stream_l362 :  Bool), but isn't accessible in the toplevel component.
    spinal.lib.Stream.connectFrom(Stream.scala:296)
    spinal.lib.Stream.$less$less(Stream.scala:100)
    spinal.tester.bug.HierarchyBug$TopLevel.<init>(HierarchyBug.scala:25)
    spinal.tester.bug.HierarchyBug$$anonfun$main$1.apply(HierarchyBug.scala:30)
    spinal.tester.bug.HierarchyBug$$anonfun$main$1.apply(HierarchyBug.scala:30)
    spinal.sim.JvmThread.run(SimManager.scala:51)

********************************************************************************
********************************************************************************

Design's errors are listed above.
SpinalHDL compiler exit stack : 

	at spinal.core.SpinalExit$.apply(Misc.scala:424)
	at spinal.core.SpinalError$.apply(Misc.scala:479)
	at spinal.core.internals.PhaseContext.checkPendingErrors(Phase.scala:175)
	at spinal.core.internals.PhaseContext.doPhase(Phase.scala:191)
	at spinal.core.internals.SpinalVhdlBoot$$anonfun$singleShot$1$$anonfun$apply$132.apply(Phase.scala:2557)
	at spinal.core.internals.SpinalVhdlBoot$$anonfun$singleShot$1$$anonfun$apply$132.apply(Phase.scala:2555)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at spinal.core.internals.SpinalVhdlBoot$$anonfun$singleShot$1.apply(Phase.scala:2555)
	at spinal.core.internals.SpinalVhdlBoot$$anonfun$singleShot$1.apply(Phase.scala:2490)
	at spinal.core.ScopeProperty$.sandbox(ScopeProperty.scala:69)
	at spinal.core.internals.SpinalVhdlBoot$.singleShot(Phase.scala:2490)
	at spinal.core.internals.SpinalVhdlBoot$.apply(Phase.scala:2485)
	at spinal.core.Spinal$.apply(Spinal.scala:387)
	at spinal.core.SpinalConfig.generate(Spinal.scala:168)
	at spinal.core.SpinalVhdl$.apply(Spinal.scala:401)
	at spinal.tester.bug.HierarchyBug$.main(HierarchyBug.scala:30)
	at spinal.tester.bug.HierarchyBug.main(HierarchyBug.scala)
```

For reference, I'm in my own fork but the latest commit from `dev` is 0f73e110

Dolu1990, 2022-08-12:
Hi,

I just tried on my side, and all went well.
May guess is that maybe the compiler plugin isn't  alive ?
Can you try running that code via sbt ?
clean
compile
runMain xxx.yyy.HierarchyBug

 ?
 
 To see if you still have the issue ?

dokleina, 2022-08-12:
I just tried it and it fixed my errors.

Thanks!

closed 2022-08-12 by dokleina