SpinalHDL/SpinalHDL #829: [VCS API] When I use VCS as a backend for emulation, it seems to have problems with Z-state warnings and VCS API does not seem to support oneFilePerComponent config

Charles-Lqx, 2022-08-21:
Hi， I encountered some error when do VCS simulation，Please take a look. Thank you.

Take a simple adder as an example：
[image]

1）Z-State Warning
Its running results are as follows：
[image]
It seems to give a Z-state warning that shouldn't be there
2）oneFilePerComponent
Use the following code to simulate：
[image]
I encountered this error after running：
[image]

Dolu1990, 2022-08-22:
Hi :)

To send the code in text format you can do : 

\`\`\`scala
Your code here
\`\`\`

It make things easier for us to reproduce.

Also which version of VCS do you have ?

Charles-Lqx, 2022-08-22:
Okay.

1) In the Z-State warning case, the example as follow:

/* scala code */
import spinal.lib._
import spinal.core._
import spinal.core.sim._

import scala.util.Random._

case class Demo() extends Component {
  val io = new Bundle {
    val a     = in UInt (8 bits)
    val b     = in UInt (8 bits)
    val ret = out UInt (9 bits)
  }

  io.ret := RegNext(io.a +^ io.b).init(U(0))
}
object demoTest extends App {
  SimConfig.withVCS(VCSFlags(elaborateFlags = List("-LDFLAGS -Wl,--no-as-needed")))
    .withFSDBWave.workspacePath(s"./simWorkSpace").allOptimisation.compile(Demo()).doSim { dut =>
    dut.io.a #= 0
    dut.io.b #= 0
    dut.clockDomain.forkStimulus(10)
    dut.io.a #= nextInt(256)
    dut.io.b #= nextInt(256)
    dut.clockDomain.waitSampling(2)
  }
}

/*end */

it's run window output log is :

/* output */

[Progress] VCS compilation started.
[Progress] VCS user environment setup.
[VLOGAN] vlogan -nc -full64 -sverilog +v2k -ntb -debug_access+all -timescale=1ns/1ps -l vlogan.log  /home/charles/IdeaProjects/SpGears/tmp/job_1/Demo.v ./rtl/__simulation_def.v
[VCS] vcs -full64 -quiet -timescale=1ns/1ps -debug_access+all -debug_acc+pp+dmptf -debug_region=+cell+encrypt -l vcs.log +vpi +vcs+initreg+random -load /home/charles/IdeaProjects/SpGears/./simWorkspace/.pluginsCachePath/vpi_vcs.so:entry_point_cb -o Demo -P /opt/synopsys/verdi201809/share/PLI/VCS/LINUX64/novas.tab /opt/synopsys/verdi201809/share/PLI/VCS/LINUX64/pli.a -LDFLAGS -Wl,--no-as-needed Demo __simulation_def
[Progress] VCS compilation done in 3.730 s
[SIMV] ./Demo 
Chronologic VCS simulator copyright 1991-2018
Contains Synopsys proprietary information.
Compiler version O-2018.09-SP2_Full64; Runtime version O-2018.09-SP2_Full64;  Aug 22 17:37 2022
Shared memory key : SpinalHDL_0_1_1944201789_-1021641580_1661161054087_-7132551111646221440
Start of simulation
*Verdi* Loading libsscore_vcs201809.so
FSDB Dumper for VCS, Release Verdi_O-2018.09-SP2, Linux x86_64/64bit, 02/21/2019
(C) 1996 - 2019 by Synopsys, Inc.
*Verdi* : Create FSDB file 'Demo.fsdb'
*Verdi* : Begin traversing the scope (Demo), layer (0).
*Verdi* : End of traversing.
*Verdi* : Flush all FSDB Files at 0 ps.
[Progress] Start Demo test simulation with seed 76217937
Warning, value 'z' at position 0 is neither '0' or '1'. value set to '0'.
Warning, value 'z' at position 0 is neither '0' or '1'. value set to '0'.
Unexpected termination of the simulation
           V C S   S i m u l a t i o n   R e p o r t 
Time: 180 ps
CPU Time:      0.340 seconds;       Data structure size:   8.0Mb
Mon Aug 22 17:37:34 2022
[Done] Simulation done in 53.592 ms

/* end */

It seems to give a Z-state warning that shouldn't be there. Because both inputs have initial values.

[... 64 lines cut ...]
	at spinal.core.sim.SimCompiled.doSim(SimBootstraps.scala:498)
	at SpGears.Other.demoTest$.delayedEndpoint$SpGears$Other$demoTest$1(SpinalHDLTest.scala:28)
	at SpGears.Other.demoTest$delayedInit$body.apply(SpinalHDLTest.scala:25)
	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.collection.immutable.List.foreach(List.scala:392)
	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
	at scala.App$class.main(App.scala:76)
	at SpGears.Other.demoTest$.main(SpinalHDLTest.scala:25)
	at SpGears.Other.demoTest.main(SpinalHDLTest.scala)

/* end */

3) My VCS version information is : 

vcs script version : O-2018.09
machine name = fpga-eda 
machine type = linux64 
machine os = Linux 5.15.0-43-generic  
The FLEXlm host ID of this machine is "000c29238993"
Compiler version = VCS-MX O-2018.09-SP2_Full64

Thank you for your reply!

Dolu1990, 2022-08-23:
Notification to @wswslzp @jijingg  :)

Charles-Lqx, 2022-08-24:
Ok! Thank you! I will continue to wait for the first question reply.

numero-744, 2022-10-03:
Hello, any updates for this issue?

wswslzp, 2022-10-04:
Hi all, sorry for my late update. I've been quite busy during this time. 

The second one should be fixed by #882 .

numero-744, 2022-10-04:
Great! What about the Z-value warnings?

Dolu1990, 2022-10-05:
Ahh the Z warnings are due to the clock domain sampling being done at the same delta cycle than the forkStimulus is done,
I pushed a fix.

So everything should be fixed now ^^

AlenLqx, 2022-10-05:
Great！Thanks.

closed 2022-10-05 by AlenLqx

Alen302, 2023-08-08:
hello! @Dolu1990 , Regarding the repair of the Z state problem of the VCS, you seem to have forgotten to merge into the new version :
https://github.com/SpinalHDL/SpinalHDL/issues/829