SpinalHDL/SpinalHDL #832: `waitRisingEdge` doesn't align with rising edge for custom clock signal

dzx-dzx, 2022-08-22:
I have a project that utilizes a Xilinx's IP, Clocking Wizard, which generates custom clock signals. The toplevel interface is something like this:
```
class myModule() extends BlackBox {
  val io = new Bundle {
    val clk                  = in Bool ()
    val ip_clk              = out Bool ()
  }
  mapCurrentClockDomain(clock = io.clk, reset = io.sys_reset)
  val IPClockDomain = new ClockDomain(clock = io.pkt_clk)
}
```
I have wrapped `ip_clk` in a `ClockDomain` and used `IPClockDomain.waitRisingEdge`. But its following logic is not executed right after it asserts, but after `clk` has also toggled(Rising or falling).

Dolu1990, 2022-08-23:
Hi,

So in the context of SpinalSim right ?

Do you have a selfcontained example to reproduce ?

dzx-dzx, 2022-08-23:
I can provide one later, may I ask if I can send it to your email?

Dolu1990, 2022-08-23:
Sure thanks

dzx-dzx, 2022-08-24:
Sent to your gmail. Thanks!

Dolu1990, 2022-08-29:
Damned Vivado, asking 150 GB of disk storage to install (i have a too old version, 2019)
I currently do not have that disk storage space.
Could you send a VCD file of the simulation ?
Thanks :)

dzx-dzx, 2022-08-29:
I failed to figure out how to convert `.wdb` to `.vcd` I'm afraid... But I think Vivado 2019 can open such format through `Flow - open static simulation`?

[image]

Dolu1990, 2022-08-29:
Ahhh you are right, that's not good.
I have no idea how the interface work between SpinalSim and vivado works. 

@name1e5s@kleinai@liuwei9
Basicaly, a blackbox delay some clock, probably via some delayed assignements, and it seems that SpinalSim doesn't see that delayed assignement as a delta cycle.

kleinai, 2022-08-29:
I think I understand the problem and I'll blame myself. I added code to read the simulation time precision and adjust that into nanoseconds. The underlying simulation can have variable time precision so it's best to query the sim and adjust accordingly.

I can see there being a problem with smaller than nanosecond steps. That was a poor assumption for me to make.

I can back that code out for now. Maybe we should think about time unit support for Spinal sims?

Dolu1990, 2022-08-31:
> Maybe we should think about time unit support for Spinal sims?

Which would change the timescale in the generated verilog ?
in which way do you mean it ?

kleinai, 2022-08-31:
Verilog time scale would be a good start in most cases. Other cases we may need to read out the sim time scale to be sure.

I can also see decoupling the sampling time scale from the fork stimulus time scale, or add units to the function.

kleinai, 2022-08-31:
I removed the extra code in the sleep function. Xilinx sims will now sleep at the kernel time scale.

Dolu1990, 2022-09-02:
So, adding a parameter to each function specifying a delay in the API as : 

```scala
def sleep(delay : Long, unit : SimTimeUnit) = sleep(delay * unit.factor) ?
```

kleinai, 2022-09-02:
It's almost that simple. We'll need to know what sleep time unit the sim is using. I'd also like to optionally set the sim time scale with something like `.withTimeScale(fs)`

There's already `TimeNumber` so maybe we could use that. Like `sleep(1 ns)` would be nice and clear.

numero-744, 2022-10-23:
What is the status of this issue?

Btw I like the `sleep(1 ns)`

dzx-dzx, 2022-10-24:
> What is the status of this issue?
> 
> Btw I like the `sleep(1 ns)`

Merged: https://github.com/SpinalHDL/SpinalHDL/pull/843
I myself haven't tested it yes, but I guess you can close it now.

closed 2022-10-24 by numero-744