SpinalHDL/SpinalHDL #833: Reg slice get not a reg error

yuqi-ali, 2022-08-23:
`
case class demo() extends Component{
  val io=new Bundle{
    val dataOutput=out UInt(64 bits) setAsReg()
  }
  io.dataOutput.setAll()
  println(io.dataOutput.isReg)
  println(io.dataOutput(31 downto 0).isReg)
}
`
Why io.dataOutput(31 downto 0) is not a reg? I get a error while using the io.dataOutput(31 downto 0) format in regif registerInWriteLogic().
@Dolu1990  @jijingg

Dolu1990, 2022-08-23:
So, xxx(y downto z) return a new signal which is assigned by the given bits, and if assigned it assign xxx range.
It's a bit special. So, technicaly speaking it is what i'm expecting, as the isReg is called on the new returned signal, which is combinatorial.

tricky tricky

yuqi-ali, 2022-08-23:
If I define a register 64 bits, while the axilite bus data width is 32 bits, It's normal in Verilog:
```
case(addr) begin
4'h0: a[31:0] <= wdata;
4'h4: a[63:32]<=wdata;
end
```
xxx(y downto z) return as combinatorial logic makes it unable to use it by this way :
```
alite.registerInWriteOnlyLogic(a[31:0])
alite.registerInWriteOnlyLogic(a[63:32])
``` 
A Little unperfect

Dolu1990, 2022-08-23:
```scala
    val rawrr = Reg(UInt(64 bits))
    rawrr(31 downto 0) := U(1, 32 bits)
    rawrr(63 downto 32) := U(2, 32 bits)

    val miaou = Reg(UInt(64 bits))
    miaou(31 downto 0).assignFromBits(B(3, 32 bits))
    miaou(63 downto 32).assignFromBits(B(4, 32 bits))

```

Works fine on my side.

What is the issue you have with 

```scala
alite.registerInWriteOnlyLogic(a[31:0])
alite.registerInWriteOnlyLogic(a[63:32])
``` 
?

When the combinatorial signal returned by xxx(y downto z) is assigned, it assign xxx not the combinatorial signal.

Do you have a self contained example ?

yuqi-ali, 2022-08-28:
```
import spinal.core._
import spinal.lib._
import spinal.lib.bus.misc.SizeMapping
import spinal.lib.bus.amba4.axilite._
import spinal.lib.bus.regif.AxiLite4BusInterface
import spinal.lib.bus.regif.AccessType._

case class demo() extends Component{
  val alite=slave(AxiLite4(12,32))
  val rawrr = Reg(UInt(64 bits))
  val aliteInterface=AxiLite4BusInterface(alite,SizeMapping(0,4 KiB))
  val reg0,reg1=aliteInterface.newReg("rawrr")
  reg0.registerInWithWriteLogic(rawrr(31 downto 0),RW,0,"rawrr low")
  reg0.registerInWithWriteLogic(rawrr(63 downto 32),RW,0,"rawrr high")
}

object demoApp extends App{
  SpinalSystemVerilog(demo())
}
you can test this demo
```

commit edbf6419 (2022-08-29): fix #833 Data.isRegOnAssign added