SpinalHDL/SpinalHDL #834: Initialize memory of bundles (literal construction?)

ColonelPhantom, 2022-08-23:
I'm trying to use a `Bundle` in a similar way to a C `struct` (i.e., as a data structure), like
```scala
case class Data() extends Bundle {
    val a = Bool
    val b = Bool
}
```
This seems to work fine, but I'm running into issues with trying to actually use it. For example, I have a memory `val memory = Mem(Data())`, but I want to assign it some default content.

In my attempt to do this, I made a helper function that creates a piece of Data, like
```scala
def construct(a: Bool, b: Bool): Data = {
    val data = Data()
    data.a = a
    data.b = b
    data
}
```
and use that to define an array, but that does not work. (My intuition is that this is because I constructed a piece of hardware, not a piece of data, is that correct?)

However, I am not sure how to construct a piece of data (`Literal`?) instead for a `Bundle`. For integers and bits there seem to be tools like `S`, `U` and `B`, but I cannot find an equivalent for Bundles.

Is there a proper way to do this, or at least a viable workaround? Or should I not use Bundles for this (yet) and twiddle bits by hand instead?

kleinai, 2022-08-23:
`Bundle`s are similar enough to `C` structs. There's a few different ways you could do about initializing your bundles.

If you want static content you would use the literal functions like `True` `False` `S` `U` `B` etc..  Case classes in Scala are required to have a parameter list (even if it's empty) where you could stick your constant values.

```scala
case class Data(a: Bool, b: Bool) extends Bundle
```

and later

```scala
val myData = Data(False, True)
```

Your example `construct` function will not work because `=` assigns the new `Bool` instance to a `val` which is not allowed in Scala. You must use `:=` to assign a value to signals.

There's also `default` which you can use like `init` inside the bundle if you're going to make the bundle a register. (I actually just found that out myself).

```scala
case class Data() extends Bundle {
  val a = Bool default(False)
  val b = Bool default(True)
}
```

and later

```scala
val myData = Reg(Data())
myData.a := True // reassign it as you want
```

ColonelPhantom, 2022-08-24:
> If you want static content you would use the literal functions like True False S U B etc.. Case classes in Scala are required to have a parameter list (even if it's empty) where you could stick your [...]
Would this still be usable in memories/registers/IO's? It seems like then the 'type' `Data(False, True)` encodes not the type `Data`, but rather a constant value.

> Your example construct function will not work because = assigns the new Bool instance to a val which is not allowed in Scala. You must use := to assign a value to signals.

Yes, you are correct. I paraphrased my code incorrectly here.
My actual code indeed uses `:=`. It seems like it actually works to create a value, however not to initialize a memory. When I try I get the error `ROM initial value should be provided from full literals value`. It seems like Spinal does not see the value returned by `construct` as a literal, but as an arbitrary signal instead.

Perhaps adding proper constructors to `Bundle`s would be useful? So that I can just type `Data.new(False, True)` (or `Data.new(a = False, b = True)` or something) and Spinal takes care of constructing it. That would seem a lot more 'functional' to me, and hopefully can be recognized by the ROM encoder. (Although I don't know if Scala is powerful enough for such metaprogramming, I have very little experience with it. Perhaps a Map would allow a similar technique with less metaprogramming?)

> There's also default which you can use like init inside the bundle if you're going to make the bundle a register. (I actually just found that out myself).

Not all my values are identical at the beginning. So I might as well leave it uninitialized and let the 'host' init the memory in that case.

I think it would be easiest to just fall back to using `Bits` for now in my case, but thanks for the help!

kleinai, 2022-08-24:
What snippets you've posted should work. For example:
```
object MyData {
  def apply(a: Bool, b: Bool): MyData = {
    val myData = MyData()
    myData.a := a
    myData.b := b
    myData
  }
}

case class MyData() extends Bundle {
  val a = Bool()
  val b = Bool()
}
```
The object MyData will provide a constructor like function to put together the bundle.
This works fine for building a memory:
```
    val memContents = Seq(
      MyData(False, False),
      MyData(False, True),
      MyData(True, True),
      MyData(False, True)
    )
    val mem = Mem(memContents)
```
`Mem` will check that all the initial values are constant. If you're providing any of these entries with a non-literal it'll fail with that message.

Dolu1990, 2022-08-25:
Avoid things as : 

```scala
case class Data(a: Bool, b: Bool) extends Bundle
```

Are considered parts of the bundle, only val defined in the body of the class.

The above code from @kleinai Worked well for me.

I would just suggest to have an annotation to the memContents, to avoid those literal signal being named, and so, being keepts in the netlist, as they are only used as temporary literals to build the rom :
 
```scala
@dontName val memContents = ...
```

ColonelPhantom, 2022-08-25:
Okay, I found out what I was doing wrong. My `construct` function did something like
```scala
def construct(flag: Bool): Data {
  val data = Data()
  data.a = flag
  data.b = !flag
  return data
}
```
and SpinalHDL does not recognize `!True` as equivalent to `False`. I fixed it by making `construct` take a Scala `Boolean` instead, and then use it like
```scala
data.a = Bool(flag)
data.b = Bool(!flag)
```
Thanks for the help! :smile: