SpinalHDL/SpinalHDL #837: Vivado: Pass flags to `xelab` and `xsim`, or allow to load custom `tcl` script

dzx-dzx, 2022-08-25:
Some IP cores from Xilinx detect a macro to alter its behavior, such as shortening simulation time. And there's a flag called `verilog_define` to apply the desired macro when compiling. But neither does SpinalHDL support passing flags(I guess?), nor can it take a custom simulating script as a workaround.

kleinai, 2022-08-25:
What sort of flags do you need and where? As part of the `xelab` command?
We could also make space for custom commands before the sim script generation is run in Vivado.

dzx-dzx, 2022-08-26:
> As part of the `xelab` command

Ah yes. `-d` specifically.

dzx-dzx, 2022-08-26:
> We could also make space for custom commands before the sim script generation is run in Vivado.

I reckon that can work too, by utilizing `set_property verilog_define`.

PR #838 (2022-08-29): XSim add custom simulation script and flag support
Closes #837
closed 2022-08-29 by kleinai