SpinalHDL/SpinalHDL #839: Axi

Readon, 2022-08-31:


Readon, 2022-08-31:
It's also necessary to carefully discuss on straight forward waitSampling in each handles, like [this](https://github.com/SpinalHDL/SpinalHDL/blob/6e128db5ac0a1a7a109439503037108f53fc850b/lib/src/main/scala/spinal/lib/bus/amba4/axi/sim/AxiMemorySim.scala#L469).

Dolu1990, 2022-08-31:
hmmm Idealy, it should not use the blocking sim API, as it is way slower than callback based API (onSampling)
I would say, it should have ben based on the StreamMonitor / StreamDriver.

closed 2022-08-31 by Dolu1990

Readon, 2022-08-31:
> 

> hmmm Idealy, it should not use the blocking sim API, as it is way slower than callback based API (onSampling) I would say, it should have ben based on the StreamMonitor / StreamDriver.

I would find time to fix this. However, I do not have write channel related test code at hand now. So it will take some time.
We might remain this issue open, and rework on the code later.

commit 49071ab0 (2022-08-31): move wait sampling while pending_reads is empty. partially fix #839
commit 77b02ea4 (2022-08-31): Merge pull request #840 from Readon/dev

move wait sampling while pending_reads is empty. partially fix #839