SpinalHDL/SpinalHDL #841: Name extraction for vals inside component func defs

huaixv, 2022-09-02:
SpinalHDL have name extraction for component vals since 1.4.0 using `spinal.idslplugin.ValCallback`,
but it doesn't handle function defs and therefore requiring a wrapper like `Area` or `Composite`.

What about extending `ValCallback` to recursively traverse function bodies and name the vals inside?

Here is a prototype for this: https://github.com/huaixv/SpinalHDL/commit/125be59ec245d468c00fe3e7861643cedaf4f492

Dolu1990, 2022-09-05:
Hi,

I just tested, and currently the situation is that the val inside a few functions produce some endless recusrive call due to the ValCallback : 

```scala
object MiaouMiaouMiaou extends App{
  SpinalVerilog(new Component{

    def func1() = {
      val x = False
    }
    func1() //One endless loop here

    def func2() = new Area {
      val a = False
    }
    val b = func2() //One endless loop here of another kind
  })
}
```

For instance the endless cal of func1 can be resolved in Bundle.scala line ~100 by adding a @dontName : 
```scala
  override def valCallback[T](ref: T, name: String): T = {
    @dontName val refs = mutable.Set[Any]()
    valCallbackOn(ref, name, refs)
    ref
  }
```

Maybe should also be added as a prefix the name of the function ?

Overall, is there issues with Area / Composite ?
In some way having automatic function naming  can be usefull, but in general, it seems to me that the Area / Composite way is better, as it allow a few things : 
- Capture the netlist structure in a object hierarchy
- Works well when there is mulitple call to the same function, as name are relative to where they are stored, not the function itself

huaixv, 2022-09-05:
> I just tested, and currently the situation is that the val inside a few functions produce some endless recusrive call due to the ValCallback :

> For instance the endless cal of func1 can be resolved in Bundle.scala line ~100 by adding a @dontname :

Ahhh, that's really strange. I can't reproduce this problem on my machine, without `@dontname`.

> Overall, is there issues with Area / Composite ?

`Area` / `Composite` is good, to provide a hierarchical way to orangize logics.
The problem is that the hierarchical name inference applies only to `val`s defined directly in the `Component`.
It won't work when logics are created outside the Component body via rework, say:
```scala
class Inner extends Component {
  var _x : Bool= null
  def setup() = {
    val x = this.rework(out(Bool()))
    _x = x
    x
  }

  def build() = this.rework {
    _x := True
  }

}

class Outer extends Component {
  val inner = new Inner()
  val x = inner.setup()
  inner.build()

  val t = out(Bool)
  t := x
}

object MiaouMiaouMiaou extends App {
  SpinalVerilog(new Outer)
}
```
Spinal will generate anon signals even if `x` in `setup` is wrapped by `Area` (unless manually call `setName`)  and many times I don't have proper `Nameable` to pass to `Composite`.

So, my "X Problem" is: how to properly name these `rework`ed signals?

Dolu1990, 2022-09-05:
You can use AreaRoot : 

```scala
  def setup() = Inner.this.rework(new AreaRoot {
    val x = out(Bool())
    _x = x
  }).x
```

AreaRoot is just an Area which has its name forced to ""  , so equivalent to new Area{ setName(""); .... }

Note that the AreaRoot will only name signal defined in the same component than itself.
Note that NaxRiscv now use a totaly different approache, using elaboration thread forks and mutex, but that's very very different.

huaixv, 2022-09-05:
Thank you very much. `AreaRoot` really helps in my case. I will also take a look at the NaxRiscv approach.

closed 2022-09-05 by huaixv