SpinalHDL/SpinalHDL #844: StreamFifoCC would accept data while reset is asserted.

Readon, 2022-09-03:
as shown below the push stream meet a reset and a fire at the same time.
The output stream would take the fired payload as input.

[image]

The code has been pushed to [my formal verification tutorial repos](https://github.com/Readon/FormalTutorials).
use below command to test it.
``` shell
    sbt "testOnly *FormalFifo*"
```

Readon, 2022-09-03:
The cause of this should be
``` verilog
always @(posedge clk) begin
    if(_zz_1) begin
      ram[_zz_ram_port] <= _zz_ram_port_1;
    end
  end
```
There is no reset condition in this generated code due to the access operation is on ram.
How about add a reset condition here?

For more details, I initiated the StreamFifoCC with the withPopBufferedReset = false to get individual reset of each clockDomain.
``` scala
val dut = FormalDut(new StreamFifoCC(cloneOf(inValue), 4, pushClock, popClock, false))
```

Readon, 2022-09-03:
Another problem is that if a totally independent reset is required in the design.
There are cases as below
[image]
After reset on push interface but pop interface, then the pushPtr would be set to 0 and broadcast to pop end.
However, the pop end has not been reseted, then the FIFO is not in empty state at the view of pop end.
So, there should be a specification for StreamFIFOCC to describe its usage.
The thing is that what is the most common usage of this facility? 
To me there is only a little knowledge on FPGA, but ASIC is out of reach.

Dolu1990, 2022-09-05:
I would say, the reset signal shouldn't be randomized as much.
Reset should be keept active more cycles to let everything having time to properly reset. It is often a requirement.

Also, i'm still not sure that having the withPopBufferedReset : Boolean = true by default was a good idea. It is like playing with fire XD

Readon, 2022-09-05:
> 

Ennn, the policy is AASD here, the reset released at clock's active edge.

For "withPopBufferedReset = true" case, 
The reset should be last at least one pop clock period time, the design works.

For "withPopBufferedReset = false" case, 
There will be two independent resets there, those two resets has to be asserted at the same time to let it work.
Resets assert independently would crash the logic inside.

Readon, 2022-09-05:
I can push those verification codes as an example for cross clock domain formal verification.

Dolu1990, 2022-09-05:
I just tried the repo, 
sbt "testOnly *FormalFifo*"

give me : 
[info] Passed: Total 0, Failed 0, Errors 0, Passed 0

Did you pushed your code ?

> Resets assert independently would crash the logic inside.

Is resets assert independently realy something to aim for ? I mean, shouldn't be a reset controller to assert / deassert things in a "right" timing ?
I don't know, they may be practical cases ?

Readon, 2022-09-05:
> 

It's on fifoCC branch

and the command should be 
sbt "testOnly *FormalFifo*"
there should be two stars around.

Dolu1990, 2022-09-05:
Ahhh ^^
good thanks

Readon, 2022-09-05:
> > Resets assert independently would crash the logic inside.
> 
> Is resets assert independently realy something to aim for ? I mean, shouldn't be a reset controller to assert / deassert things in a "right" timing ? I don't know, they may be practical cases ?

I think it's required by a special case, not the common one. 
That's why I just gives an assumption on that. 
I think the design is good enough, the verification just give a boundary on the usage of the design.

Dolu1990, 2022-09-05:
So, what's about : 

"""When push side reset is asserted, the pop side may be corrupted until the reset get propagated through (3 pop cycles)"""

I'm saying get corrupted, because the counter get reset, which violate the gray counter properties of one transition at the time.
Does it sound good ?

Readon, 2022-09-05:
> So, what's about :
> 
> """When push side reset is asserted, the pop side may be corrupted until the reset get propagated through (3 pop cycles)"""
> 
> I'm saying get corrupted, because the counter get reset, which violate the gray counter properties of one transition at the time. Does it sound good ?

It still depends,
For "withPopBufferedReset = true" case,
The pop end would reset when there is a reset rising. 
However, the reset still should last longer than one pop clock period.

For "withPopBufferedReset = false" case,
There will be no reset propagation, it would actually corrupt the counting mechanism.
So a pop reset is a must. I did not verify on reset active at different time. That might be a rare case.

Readon, 2022-09-05:
> For "withPopBufferedReset = false" case, There will be no reset propagation, it would actually corrupt the counting mechanism. So a pop reset is a must. I did not verify on reset active at different time. That might be a rare case.

In this case, a reset at only one end will corrupt the counter and cause the design to output the wrong data.

closed 2022-09-06 by Readon