SpinalHDL/SpinalHDL #845: Is there any possibility to get the instance drive one signal.

Readon, 2022-09-04:
There are cases that signal is generated by hidden object, as
``` scala
    val x = BufferCC(y)
```
where x can be reference through top level instance, however the object created by BufferCC is not.
This prevent formal verification in prove mode to check the status of registers creeated by BufferCC.

If there is a function like x.getDriveInstance to get the object created by BufferCC then the check would be possible.

Dolu1990, 2022-09-05:
There is xxx.getSingleDriver, works for BaseType

Readon, 2022-09-05:
> There is xxx.getSingleDriver, works for BaseType

closed 2022-09-05 by Readon

Dolu1990, 2022-09-05:
There is also getDrivingReg which will walk through single driver combinatorial stuff until it find the register

Readon, 2022-09-05:
> There is also getDrivingReg which will walk through single driver combinatorial stuff until it find the register

That is enough temperarily, thanks.

Dolu1990, 2022-09-05:
Ahh one thing you can do, which is kind of a workaround, is to retreive stuff by their name : 

myComponent.reflectBaseType("myRegisterName") it will return the base type in the netlist which has that name.