SpinalHDL/SpinalHDL #846: Call other FSM's isActive method in whenIsActive results in NO DRIVER error

wmj0607, 2022-09-05:
using v1.7.2 with scala 2.12.16, the example code:
```scala
object Test {
  def main(args: Array[String]) {SpinalConfig(mode = Verilog).generate{new Component{
    val o = out(Bool())
    val fsm1 = new StateMachine{
      val s1 = makeInstantEntry()
      val s2 = new State
      s1.whenIsActive{ goto(s2) }
      s2.whenIsActive{ goto(s1) }
    }
    val fsm2 = new StateMachine{
      val s1 = makeInstantEntry()
      val s2 = new State
      s1.whenIsActive{ when(fsm1.isActive(fsm1.s2)){goto(s2)} } //error
      s2.whenIsActive{ goto(s1) }
    }
    o := fsm2.isActive(fsm2.s2)
  }}}
}
```
causes NO DRIVER error on when condition in the error line. 

A simple way to avoid this is create a val for isActive at somewhere outside whenIsActive (it's fine in fsm2).

Dolu1990, 2022-09-05:
Hi,

Yes it is "normal", the code in the whenIsActive is only called when fsm2 is being builded, which happen after fsm1.
Issue is that the logic driving the fsm1.isActive(xxx) is only genereted when fsm1 is being builded. Futher cals to isActive will be ignored.

Dolu1990, 2022-09-05:
Fixed

commit 020da4e2 (2022-09-05): Fix #846 fsm.isXxx can now be called after the FSM was builded.
closed 2022-09-05 by Dolu1990