SpinalHDL/SpinalHDL #849: How to connect fifo with BusIf?

Readon, 2022-09-09:
Normally, memory mapped interface support register, FIFO and ram. 
The ram can directly collect by cross bar.
But existing StreamFIFO is not addressable.
How to connect BusIf with FIFO is a problem, it's best to add an example to demonstrate the usage.

Dolu1990, 2022-09-16:
I'm not sure what is the best way, i haven't use the API much.
@jijingg

jijingg, 2022-09-19:
hi @Readon @Dolu1990 , the FIFO and RAM interface of BusIf have not complete yet. 
I haven't figured out whether instantiate FIFO to regbank.v Internal or direct output fifo port is better

Readon, 2022-09-19:
> hi @Readon @Dolu1990 , the FIFO and RAM interface of BusIf have not complete yet. I haven't figured out whether instantiate FIFO to regbank.v Internal or direct output fifo port is better

I think leave FIFO outside of regbank is better for decoupling.
RAM can be connect through independent bus through crossbar.

Dolu1990, 2022-09-19:
Hmm maybe before thinking fifo we should think stream ? 
For example (but not necessarly a good example) : 
https://github.com/SpinalHDL/SpinalHDL/blob/e1e5961debaa029d7efa0cc8896048752225d57f/lib/src/main/scala/spinal/lib/bus/misc/BusSlaveFactory.scala#L529

Readon, 2024-10-05:
closed by #1409

closed 2024-10-05 by Readon