SpinalHDL/SpinalHDL #851: [Bug --update SpinalHDL--Help]HIERARCHY VIOLATION : when using StreamJoin

dreamflyings, 2022-09-11:
When I upgraded the spinalhdl library to the latest version more than three months ago, I encountered an unexpected error@ Dolu1990 can you find time to look and patch it as soon as possible? Thank you very much! The demo is as follows:
```scala
object StreamJoinPlay extends App{

  // todo it's wrong
  SpinalConfig().generateSystemVerilog(new Module{
    val s0 = slave Stream(UInt(8 bits))
    val s1 = slave Stream(Fragment(UInt(8 bits)) )
    val s  = master Stream(TupleBundle2(UInt(8 bits), Fragment(UInt(8 bits)) ) )
    case class TestM(left:Stream[UInt], right:Stream[Fragment[UInt]]) extends ImplicitArea[Stream[TupleBundle2[UInt, Fragment[UInt]]]] {
      val sout = Stream(TupleBundle2(UInt(8 bits), Fragment(UInt(8 bits)) ) )
      val so = StreamJoin(left, right)
      val sin = so.queue(1)
      sout.arbitrationFrom(sin)
      sout._1           := sin._1
      sout._2.fragment  := sin._1 + sin._2.fragment
      sout._2.last      := sin._2.last
      override def implicitValue: Stream[TupleBundle2[UInt, Fragment[UInt]]] = sout
    }
    s << TestM(s0, s1)
  })

  // todo it's right
  SpinalConfig().generateSystemVerilog(new Module{
    val s0 = slave Stream(UInt(8 bits))
    val s1 = slave Stream(Fragment(UInt(8 bits)) )
    val s  = master Stream(TupleBundle2(UInt(8 bits), Fragment(UInt(8 bits)) ) )
    case class TestB() extends Bundle {
      val sd0 = UInt(8 bits)
      val sd1 = Fragment(UInt(8 bits))
    }
    case class TestM(left:Stream[UInt], right:Stream[Fragment[UInt]]) extends ImplicitArea[Stream[TupleBundle2[UInt, Fragment[UInt]]]] {
      val sout = Stream(TupleBundle2(UInt(8 bits), Fragment(UInt(8 bits)) ) )
      val so = StreamJoin(left, right)
      val tmpb = TestB()
      tmpb.sd0 := so._1
      tmpb.sd1 := so._2
      val soo= so.translateWith(tmpb)
      val sin = soo.queue(1)
      sout.arbitrationFrom(sin)
      sout._1           := sin.sd0
      sout._2.fragment  := sin.sd0 + sin.sd1.fragment
      sout._2.last      := sin.sd1.last
      override def implicitValue: Stream[TupleBundle2[UInt, Fragment[UInt]]] = sout
    }
    s << TestM(s0, s1)
  })
}

```

kleinai, 2022-09-11:
Can you share an error message from your example?  I ran this myself on dev and found that your "wrong" code generated fine and your "right" code had this error:

```
 Error detected in phase PhaseCheck_noLatchNoOverride
********************************************************************************
********************************************************************************

NO DRIVER ON (toplevel/??? :  Bool), defined at
    spinal.lib.Fragment.<init>(Fragment.scala:464)
    spinal.lib.FragmentFactory.apply(Fragment.scala:8)
    StreamJoinPlay2$$anon$2$TestB.<init>(temp.scala:37)
    StreamJoinPlay2$$anon$2$TestM.<init>(temp.scala:42)
    StreamJoinPlay2$$anon$2.<init>(temp.scala:54)
    StreamJoinPlay2$.$anonfun$new$2(temp.scala:31)
    spinal.sim.JvmThread.run(SimManager.scala:51)
```

Oddly you'd need to assign `tmpb.sd1.last` directly rather than letting the bundle assignment take care of it.  Seems like a bug..?

```
      val tmpb = TestB()
      tmpb.sd0 := so._1
      tmpb.sd1 := so._2
      tmpb.sd1.last := so._2.last
```

dreamflyings, 2022-09-12:
> Can you share an error message from your example? I ran this myself on dev and found that your "wrong" code generated fine and your "right" code had this error:
>
> ```
>  Error detected in phase [...]

First's Wrong message:
```
[Runtime] SpinalHDL dev    git head : 020da4e21c6d2114872a623780763f818f346e6e
[Runtime] JVM max memory : 3916.0MiB
[Runtime] Current date : 2022.09.12 08:43:02
[Progress] at 0.000 : Elaborate components
[Progress] at 0.442 : Checks and transforms

**********************************************************************************************
[Warning] Elaboration failed (5 errors).
          Spinal will restart with scala trace to help you to find the problem.
**********************************************************************************************

[Progress] at 0.455 : Elaborate components
[Progress] at 0.460 : Checks and transforms
Exception in thread "main" spinal.core.SpinalExit: 

 Error detected in phase PhaseCheckHiearchy
********************************************************************************
********************************************************************************

HIERARCHY VIOLATION : (toplevel/[StreamFifo]/??? :  UInt[8 bits]) is used to drive the (toplevel/??? :  UInt[8 bits]) := (toplevel/[StreamFifo]/??? :  UInt[8 bits]) statement, but isn't readable in the toplevel component
    demo.StreamJoinPlay$$anonfun$1$$anon$1$TestM.<init>(StreamJoinPlay.scala:17)
    demo.StreamJoinPlay$$anonfun$1$$anon$1.<init>(StreamJoinPlay.scala:22)
    demo.StreamJoinPlay$$anonfun$1.apply(StreamJoinPlay.scala:8)
    demo.StreamJoinPlay$$anonfun$1.apply(StreamJoinPlay.scala:8)
    spinal.sim.JvmThread.run(SimManager.scala:51)

********************************************************************************
********************************************************************************

HIERARCHY VIOLATION : (toplevel/[StreamFifo]/??? :  UInt[8 bits]) is used to drive the (toplevel/??? :  UInt[? bits]) := (UInt + UInt)[8 bits] statement, but isn't readable in the toplevel component
    demo.StreamJoinPlay$$anonfun$1$$anon$1$TestM.<init>(StreamJoinPlay.scala:18)
    demo.StreamJoinPlay$$anonfun$1$$anon$1.<init>(StreamJoinPlay.scala:22)
    demo.StreamJoinPlay$$anonfun$1.apply(StreamJoinPlay.scala:8)
    demo.StreamJoinPlay$$anonfun$1.apply(StreamJoinPlay.scala:8)
    spinal.sim.JvmThread.run(SimManager.scala:51)

********************************************************************************
********************************************************************************

HIERARCHY VIOLATION : (toplevel/[StreamFifo]/??? :  UInt[8 bits]) is used to drive the (toplevel/??? :  UInt[? bits]) := (UInt + UInt)[8 bits] statement, but isn't readable in the toplevel component
    demo.StreamJoinPlay$$anonfun$1$$anon$1$TestM.<init>(StreamJoinPlay.scala:18)
    demo.StreamJoinPlay$$anonfun$1$$anon$1.<init>(StreamJoinPlay.scala:22)
    demo.StreamJoinPlay$$anonfun$1.apply(StreamJoinPlay.scala:8)
    demo.StreamJoinPlay$$anonfun$1.apply(StreamJoinPlay.scala:8)
    spinal.sim.JvmThread.run(SimManager.scala:51)

********************************************************************************
********************************************************************************

HIERARCHY VIOLATION : (toplevel/[StreamFifo]/??? :  Bool) is used to drive the (toplevel/??? :  Bool) := (toplevel/[StreamFifo]/??? :  Bool) statement, but isn't readable in the toplevel component
    demo.StreamJoinPlay$$anonfun$1$$anon$1$TestM.<init>(StreamJoinPlay.scala:19)
    demo.StreamJoinPlay$$anonfun$1$$anon$1.<init>(StreamJoinPlay.scala:22)
    demo.StreamJoinPlay$$anonfun$1.apply(StreamJoinPlay.scala:8)
    demo.StreamJoinPlay$$anonfun$1.apply(StreamJoinPlay.scala:8)
    spinal.sim.JvmThread.run(SimManager.scala:51)

********************************************************************************
********************************************************************************

Design's errors are listed above.
SpinalHDL compiler exit stack : 

	at spinal.core.SpinalExit$.apply(Misc.scala:424)
	at spinal.core.SpinalError$.apply(Misc.scala:479)
	at spinal.core.internals.PhaseContext.checkPendingErrors(Phase.scala:175)
	at spinal.core.internals.PhaseContext.doPhase(Phase.scala:191)
	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2$$anonfun$apply$137.apply(Phase.scala:2714)
	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2$$anonfun$apply$137.apply(Phase.scala:2712)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2.apply(Phase.scala:2712)
	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2.apply(Phase.scala:2648)
	at spinal.core.ScopeProperty$.sandbox(ScopeProperty.scala:69)
	at spinal.core.internals.SpinalVerilogBoot$.singleShot(Phase.scala:2648)
	at spinal.core.internals.SpinalVerilogBoot$.apply(Phase.scala:2643)
	at spinal.core.Spinal$.apply(Spinal.scala:393)
	at spinal.core.SpinalConfig.generateSystemVerilog(Spinal.scala:171)
	at demo.StreamJoinPlay$.delayedEndpoint$demo$StreamJoinPlay$1(StreamJoinPlay.scala:8)
	at demo.StreamJoinPlay$delayedInit$body.apply(StreamJoinPlay.scala:5)
	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.collection.immutable.List.foreach(List.scala:392)
	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
	at scala.App$class.main(App.scala:76)
	at demo.StreamJoinPlay$.main(StreamJoinPlay.scala:5)
	at demo.StreamJoinPlay.main(StreamJoinPlay.scala)

```

dreamflyings, 2022-09-12:
Second's Right message:
```
[Runtime] SpinalHDL dev    git head : 020da4e21c6d2114872a623780763f818f346e6e
[Runtime] JVM max memory : 3916.0MiB
[Runtime] Current date : 2022.09.12 08:50:30
[Progress] at 0.000 : Elaborate components
[Progress] at 0.386 : Checks and transforms
[Progress] at 0.418 : Generate Verilog
[Warning] 14 signals were pruned. You can call printPruned on the backend report to get more informations.
[Done] at 0.449
[Warning] Pruned wire detected : (toplevel/streamFifo_1/io_push_m2sPipe_payload_sd1_fragment :  UInt[8 bits])
[Warning] Pruned wire detected : (toplevel/streamFifo_1/io_push_rData_sd1_fragment :  UInt[8 bits])
[Warning] Pruned wire detected : (toplevel/streamFifo_1/io_push_payload_sd0 : in UInt[8 bits])
[Warning] Pruned wire detected : (toplevel/streamFifo_1/io_push_rData_sd1_last :  Bool)
[Warning] Pruned wire detected : (toplevel/streamFifo_1/io_push_payload_sd1_last : in Bool)
[Warning] Pruned wire detected : (toplevel/streamFifo_1/io_availability : out UInt[1 bits])
[Warning] Pruned wire detected : (toplevel/streamFifo_1/io_push_payload_sd1_fragment : in UInt[8 bits])
[Warning] Pruned wire detected : (toplevel/streamFifo_1/io_pop_payload_sd1_last : out Bool)
[Warning] Pruned wire detected : (toplevel/streamFifo_1/io_push_m2sPipe_payload_sd1_last :  Bool)
[Warning] Pruned wire detected : (toplevel/streamFifo_1/io_pop_payload_sd0 : out UInt[8 bits])
[Warning] Pruned wire detected : (toplevel/streamFifo_1/io_push_m2sPipe_payload_sd0 :  UInt[8 bits])
[Warning] Pruned wire detected : (toplevel/streamFifo_1/io_occupancy : out UInt[1 bits])
[Warning] Pruned wire detected : (toplevel/streamFifo_1/io_pop_payload_sd1_fragment : out UInt[8 bits])
[Warning] Pruned wire detected : (toplevel/streamFifo_1/io_push_rData_sd0 :  UInt[8 bits])
```

Dolu1990, 2022-09-12:
@dreamflyings Hi

Hmm it maybe possible that you have some missconfigured sbt or broken compilation cache in scala.
Did you tried to do sbt clean in SpinalHDL and your project ?

Because it maybe just be the SpinalHDL Scala compiler plugin not being alive.

Can you share your sbt buildscripts ?

Dolu1990, 2022-09-14:
@dreamflyings Do you have a "simple" self contained example to reproduce the issue with mergeRTLSource ?

(I never used the feature)

dreamflyings, 2022-09-14:
> @dreamflyings Do you have a "simple" self contained example to reproduce the issue with mergeRTLSource ?
> 
> (I never used the feature)

@Dolu1990 hi~
After the problem is purified, it seems that the reason is also clear.
```scala
import spinal.core._

class BlackBoxMerge() extends BlackBox{
 val a = in Bool()
 val b = in Bool()
 val c = out Bool()
}
class BlackBoxMegeFixture() extends Component {
 val a = in Bool()
 val b = in Bool()
 val c = out Bool()
 val mod = new BlackBoxMerge()
 mod.a := a
 mod.b := b
 c := mod.c
}
object BlackBoxMerge_Gen extends App{
 new {
 val rep = SpinalSystemVerilog(SpinalConfig()){new BlackBoxMegeFixture}
 rep.blackboxesSourcesPaths.add(rep.generatedSourcesPaths.head)
 rep.mergeRTLSource("BlackBoxMegeFixture")
 }
 new {
 val rep = SpinalSystemVerilog(SpinalConfig()){new BlackBoxMegeFixture().setDefinitionName("BlackBoxMegeFixture_1")}
 rep.blackboxesSourcesPaths.add(rep.generatedSourcesPaths.head)
// rep.mergeRTLSource("BlackBoxMegeFixture")
 }
 new {
 val rep = SpinalSystemVerilog(SpinalConfig()){new BlackBoxMegeFixture().setDefinitionName("BlackBoxMegeFixture_2")}
 rep.blackboxesSourcesPaths.add(rep.generatedSourcesPaths.head)
 rep.mergeRTLSource("BlackBoxMegeFixture_2")
 }
}

```

Dolu1990, 2022-09-16:
Ahhh so the issue is when you override a file with mergeRTLSource, and that file is in the file list ?

Right ?

dreamflyings, 2022-09-19:
> Ahhh so the issue is when you override a file with mergeRTLSource, and that file is in the file list ?
> 
> Right ?

That's right! However, previous versions are possible. I was thinking, should we be compatible with the previous library code?

Dolu1990, 2022-09-19:
Ahh should be fixable, right. will check what changes happend

dreamflyings, 2022-09-19:
> Ahh should be fixable, right. will check what changes happend

Great praise!

Dolu1990, 2022-09-20:
@dreamflyings can you try with the dev branch ? i just pushed a fix, let's me know if it is all good with it :)

dreamflyings, 2022-09-20:
> @dreamflyings can you try with the dev branch ? i just pushed a fix, let's me know if it is all good with it :)
I update to the latest development version.
It is all good with the dev branch!
Thx~

closed 2022-09-20 by dreamflyings