SpinalHDL/SpinalHDL #852: [Bug] Lib\sim StreamDriver FSM

dokleina, 2022-09-11:
StreamDriver's `fsm` method recursively calls itself:

```scala
def fsm(): Unit = {
    state match{
      case 0 => {
        if (delay == 0) {
          state += 1
          fsm() // *** Called here ***
        } else {
          delay -= 1
        }
      }
      case 1 => {
        if(driver(stream.payload)){
          stream.valid #= true
          state += 1
        }
      }
      case 2 => {
        if(stream.ready.toBoolean){
          stream.valid #= false
          stream.payload.randomize()
          delay = transactionDelay()
          state = 0
          fsm() // *** Called here ***
        }
      }
    }
  }
```

This causes multiple runs of the FSM within one clock. This leads to inconsistent results in the payload value given to the driver function and the payload in the stream.

Removing the recursive calls appears to fix the problem.

andreasWallner, 2022-09-11:
To me the code looks good: the `state=0` before the recursive call will make sure that the other code paths are entered during the recursive calls (to e.g. allow valid to go to true in the first cycle in which ready is true).

There might be a misunderstanding in how the driver function is supposed to work: the payload is actually not passed to the driver function. Actually the `driver` function is required to **set** the correct value and return true if a valid value was set (or false if e.g. none is available at the moment) - I don't see a reason for it to read the payload value.
To record the values that the driver generates you'll want to use the `StreamMonitor`.
For an example e.g. have a look here: https://github.com/SpinalHDL/SpinalWorkshop/blob/5e3d26de253b90784fa8eb57190280f8bf73c7cf/src/test/scala/workshop/stream/StreamUnitTester.scala

dokleina, 2022-09-11:
@andreasWallner 

That's exactly what I was doing wrong.

I modified my sim to follow that example and everything is working as expected.

Thanks!

closed 2022-09-11 by dokleina