SpinalHDL/SpinalHDL #853: Strange behavior of the StreamFragmentWidthAdapter

jonnykl, 2022-09-16:
The behavior of the `StreamFragmentWidthAdapter` differs from what I'm expecting.

I want to convert a Stream of n bits into a Stream of a multiple of n bits. The current implementation ignores the `last` flag of the input stream if it's not the last element of the output data. This way the `last` flag of the input stream is very likely to be missed. I think it should be ensured that this can never happen. The solution would be to pad the output data somehow when the `last` flag of the input stream is set. Then you could also set the `last` flag of the output stream. This way it is not possible to miss the `last` flag of the input stream.

The following code demonstrates the issue of the missed `last` flag:

```scala
import spinal.core._
import spinal.core.sim._
import spinal.lib._
import spinal.lib.sim._

object FooSim {
  def main(args: Array[String]): Unit = {
    SimConfig.withFstWave.compile({
      new Component() {
        val io = new Bundle {
          val inputStream = slave Stream(Fragment(Bits(1 bits)))
          val outputStream = master Stream(Fragment(Bits(8 bits)))
        }

        StreamFragmentWidthAdapter(io.inputStream, io.outputStream, padding=true)
      }
    }).doSim(seed=42) { dut =>
      val clockPeriod = 10
      dut.clockDomain.forkStimulus(clockPeriod)

      dut.io.outputStream.ready #= true

      var cnt = 0
      StreamDriver(dut.io.inputStream, dut.clockDomain) { data =>
        data.fragment #= 0
        data.last #= cnt == 5

        cnt += 1
        cnt <= 8
      }

      StreamMonitor(dut.io.outputStream, dut.clockDomain) { data =>
        println(s"data=${data.fragment.toBigInt.formatted("%08x")} last=${data.last.toBoolean}")
      }

      dut.clockDomain.waitSampling(100)
    }
  }
}
```

Dolu1990, 2022-09-19:
Got it fixed with : 
data=2a last=true
data=55 last=false
data=d5 last=true   <- buffer isn't cleared, on purpose to save are
You need to set the earlyLast parameter to enable the required hardware.

Dolu1990, 2022-09-19:
By the way, thankgs for the nice example with expected outputs :D

closed 2022-09-19 by Dolu1990

jonnykl, 2022-09-19:
Seems to be not working for me. This is the output of my example (only added the `earlyLast` argument):
```
[info] data=aa last=true
[info] data=aa last=false
[info] data=aa last=true
```

I'm also not sure if it's clever to not clear the buffer. If you would know how much data will be sent/received, you wouldn't need the `last` flag. So there should be a way to indicate which bits are valid (in the case `last=True`). Maybe the adapter could output an additional `valid` bit vector. Then it should be OK to not clear the buffer (user can do it based on the `valid` bit vector - if needed).

Dolu1990, 2022-09-20:
Ahhh having a bit mask, right could be a thing too.

Added as dut.adapter.dataMask in the example bellow

```scala
object FooSim {
  def main(args: Array[String]): Unit = {
    SimConfig.withFstWave.compile({
      new Component() {
        val io = new Bundle {
          val inputStream = slave Stream(Fragment(Bits(1 bits)))
          val outputStream = master Stream(Fragment(Bits(8 bits)))
        }

        val adapter = StreamFragmentWidthAdapter(io.inputStream, io.outputStream, padding=true, earlyLast= true)
        adapter.dataMask.simPublic()
      }
    }).doSim(seed=42) { dut =>
      val clockPeriod = 10
      dut.clockDomain.forkStimulus(clockPeriod)

      dut.io.outputStream.ready #= true

      var cnt = 0
      StreamDriver(dut.io.inputStream, dut.clockDomain) { data =>
        data.fragment #= cnt % 2
        data.last #= cnt == 6 || cnt == 15

        cnt += 1
        cnt <= 16
      }

      StreamMonitor(dut.io.outputStream, dut.clockDomain) { data =>
        println(s"data=${data.fragment.toBigInt.formatted("%02x")} last=${data.last.toBoolean} mask=${dut.adapter.dataMask.toBigInt.toString(16)}")
      }

      dut.clockDomain.waitSampling(100)
    }
  }
}
``` 

data=2a last=true mask=7f
data=55 last=false mask=ff
data=d5 last=true mask=1

jonnykl, 2022-09-21:
Looks good, thanks :)

commit e6b2c67b (2022-09-19): Fix #853 Add StreamFragmentWidthAdapter earlyLast parameter