SpinalHDL/SpinalHDL #860: how to pull data or push data to Top

jijingg, 2022-09-19:
hi @Dolu1990 

how to pull data from or push data to   another hierarachy as wire .
- case1:  pullDataFromTopPort as input
- case2:  pullDataFromTopHier as input 
- case3:  pushDataToTopPort as output 
[image]

something like 
```scala
class Top extends Component{
val sub1 = new Sub1()

val synrst = Bool()
}
class Sub1 extends Component{
val sub2 = new Sub2()
}
class Sub2 extends Component{
val case1 = pullDataFromTopPort(Bits(8 bit), "data0")    
val case2 = pullDataFromTop("synrst")
val data = Bits(8 bit)
val case3 = pushDataTopTopPort(data)
}
```

i know how to implement case1 , but  I'm stuck with case2 pullDataFrom an exits internal signals , 

because i can add add get io signals  with getAllIo method , but i don't know how to get signal internal in top. 

Case 3 is still very difficult. I don't know how to push the existing signals of a sub module

Do you have any good suggestions I want to use these features to implement dft and scan signals

Dolu1990, 2022-09-19:
Hi jijingg,

>  i can add add get io signals with getAllIo method

I think that's one thing you should not do as directly, and instead rely on more regular API.

Here is some code, as i understand your intend are : 

```scala
object Miaou extends App{
  def pullDataFromTopPort[T <: Data](hardType : HardType[T], name : String) : T = {
    val ret = hardType()
    val top = Component.current.parents().head
    val topIo = top.rework(in(hardType())).setName(name)
    topIo.pull(propagateName = true)
  }

  def pullDataFromTop(name : String) : BaseType = {
    val top = Component.current.parents().head
    val topIo = top.reflectBaseType(name)
    topIo.pull(propagateName = true)
  }

  def pushDataTopTopPort[T <: Data](that : T) : T = {
    val top = Component.current.parents().head
    top.rework {
      val topPulled = that.pull(propagateName = true)
      out(CombInit(topPulled)).setCompositeName(topPulled)
    }
  }

  class Top extends Component{
    val synrst = Bool()

    val sub1 = new Sub1()
  }
  class Sub1 extends Component{
    val sub2 = new Sub2()
  }
  class Sub2 extends Component{
    val case1 = pullDataFromTopPort(Bits(8 bit), "data0")
    val case2 = pullDataFromTop("synrst")
    val data = Bits(8 bit); data := 0
    val case3 = pushDataTopTopPort(data)
  }

  SpinalVerilog(new Top())
}
```

Note synrst has to be defined before sub1  in the toplevel, it is ok with what you need to do ? or that is your main issue ?

jijingg, 2022-09-19:
great,  i have not notice that `top.reflectBaseType(name)` can found wire in the Component. 

this three function  exactly what i want , thanks :+1:

Dolu1990, 2022-09-19:
Note you can also give toplevel signals references to subcomponents via the sub component class constructor, to avoid using reflectBaseType.
As long as you use those toplevel signals reference to call the pull function it is all good.

dreamflyings, 2022-09-19:
> Note you can also give toplevel signals references to subcomponents via the sub component class constructor, to avoid using reflectBaseType. As long as you use those toplevel signals reference to call the pull function it is all good.

It's great!
@Dolu1990 
Can these three functions, as well as other similar auxiliary functions, be updated to the library? As far as I know, the xsim simulation is not perfect at present, and the internal signal of the module does not support the simPublic function. The purpose of simulation can also be achieved with these three functions.

Dolu1990, 2022-09-19:
@dreamflyings Sure sure, we can add those function in the API.
As part of the object Data (https://github.com/SpinalHDL/SpinalHDL/blob/8dca501481767fea74cc836d186c0ce2931bd78f/core/src/main/scala/spinal/core/Data.scala#L150)
Or as part of a new object Pull ?

But maybe not the one using the reflectBaseType, as it is kind of realy hacky, instead we could provide a reference to the toplevel signal.

dreamflyings, 2022-09-19:
> @dreamflyings Sure sure, we can add those function in the API. As part of the object Data ( [...]

(⸝⸝•‧̫•⸝⸝) ♡
Logically speaking, it is reasonable to put it into the Data object. However, we should also consider scalability and easy to remember. It depends on your comprehensive consideration.

Dolu1990, 2022-09-20:
Added in https://github.com/SpinalHDL/SpinalHDL/commit/ad1096d0009139902b749605e0fd1af95295f2a6

Can be used as : 

```scala
object Miaou extends App{
  class Top extends Component{
    val sub1Inst = new Sub1()
  }
  class Sub1 extends Component{
    val sub2Inst = new Sub2()
  }
  class Sub2 extends Component{
    val getIn  = Pull.driveFromTopInput(Bits(8 bits))
    val getOut = Pull.toTopOutput(Bits(8 bits))
    getOut := getIn
  }

  SpinalVerilog(new Top())
}
```

will generate as toplevel io : 

```
module Top (
  input      [7:0]    toplevel_sub1Inst_sub2Inst_getIn,
  output     [7:0]    toplevel_sub1Inst_sub2Inst_getOut
);
```

closed 2022-09-20 by Dolu1990