SpinalHDL/SpinalHDL #863: Add option to turn `OUT OF RANGE CONSTANT` error into a warning?

jonnykl, 2022-09-19:
Sometimes it is OK to compare against a constant outside of the range of the signal. Example: Component with multiple configurations. The component has a `n`-bit `Counter` and the counter value will be compared with a given configuration variable (`counter.value < config_var`). If the configuration variable is `2^n`, SpinalHDL will throw the `OUT OF RANGE CONSTANT` error. I think it would be nice if it's possible to turn that error into a warning. Either using a global option or somehow explictly for every comparison.

Dolu1990, 2022-09-19:
Sure, why not, global option as a emergency case, and should be normaly dealed using by using a by comparaison allowance.

Maybe we could apply the allowance on the literal ?

jonnykl, 2022-09-21:
> Maybe we could apply the allowance on the literal ?

I think that would be the best solution. I think it should be possible to define an implicit class with a `allowOutOfRange` method. This way something like the following should be possible:

```scala
val n = 8
val counter = Counter(n)
when(counter < n.allowOutOfRange) {
  // ...
}
```

closed 2022-09-21 by Dolu1990

jonnykl, 2022-09-21:
Nice, thanks :)

Now `when(counter < n allowOutOfRangeLiterals) {}` or even `when(n > counter allowOutOfRangeLiterals) {}` is possible which looks somewhat prettier :+1:

Dolu1990, 2022-09-21:
Hooooo yes right, i didn't thinked about this ^^

Thanks ^^

commit 56400992 (2022-09-21): fix #863  Add allowOutOfRangeLiterals(my4bits === 42) to skip error. Also add SpinalConfig(allowOutOfRangeLiterals = true)  to apply it everywhere