SpinalHDL/SpinalHDL #866: Feature: add `whenNone` to represent empty `WhenContext`?

tanhongze, 2022-09-20:
Sometimes, I want to generate `when` conditionally.
We could have a special object `whenNone` with type `WhenContext` to represent empty context:
```
object whenNone extends WhenContext{
  def elsewhen(cond:Bool)(action: => Unit):WhenContext = {
    when(cond){
      action
    }
  }
  def otherwise(action: => Unit){
    action
  }
}
```
Then codes could be like:
```
var ctx = whenNone
if(flag){
  ctx.elsewhen(cond1){
    assignment1
  }
}
ctx.elsewhen(cond2){
    assignment2
}
``` 
Although we could reorder the assignments and drop all `elsewhen`, I think this could be a smart and natural way to manage conditional assignments.

Dolu1990, 2022-09-21:
Hi,

Yes, why not, it can be usefull ^^
Maybe would have to think about another naming for whenNone, as None as quite some meanings.

Maybe whenBuilder ? or something in that kind ?

Maybe the object do not need to extends WhenContext, and instead create and maintine a WhenContext internaly, 
With the solution you propose, don't you need to always reassign ctx ? ex : 
ctx = ctx.elsewhen(cond2){
    assignment2
}

tanhongze, 2022-09-21:
Sounds good :)
Let it works like:
```
val builder = whenBuilder
if(flag){
  builder.elsewhen(cond1){
    assignment1
  }
}
builder.elsewhen(cond2){
    assignment2
}
```

Dolu1990, 2022-09-21:
The build could even hide the first elsewhen : 

```scala
val builder = whenBuilder
if(flag){
  builder.when(cond1){
    assignment1
  }
}
builder.elsewhen(cond2){
    assignment2
}
```

Just for the look, elsewhen and when would be implemented the same.

tanhongze, 2022-09-22:
#868

closed 2022-09-22 by Dolu1990