SpinalHDL/SpinalHDL #872: A common naming policy for formal verification related facilities.

Readon, 2022-09-25:
As the formal verification facilities' base code getting large, it's necessary to have a commonly adopted policy for those code.
What I can conclude now is:
1. There could be withAsserts function for Components for common internal assertions might be helpful for "prove".
2. There could be withMasterAsserts, withSlaveAsserts for a type of bus, and note them as "withXXXAsserts" below. So that the interface defined in "io" can be asserted by most commonly required assertion.
3. There could be withXXXAssumes and withXXXCovers for buses, so that common assumption and coverage for bus is provided.
4. The special case is the Stream interface, where only withAsserts for master interface of DUT component, and withAssumes for slave interface. withCovers can be applied to both master and slave.

So that a formal verification code can be a bundle of those common features with some small assertions and assumptions for special usage.

Dolu1990, 2022-09-26:
withAsserts should be also fine for regular simulation too. Else has to be withFormalAsserts i think ?
I think 2 3 4 5 are good ^^

Readon, 2022-09-26:
> withAsserts should be also fine for regular simulation too. Else has to be withFormalAsserts i think ? I think 2 3 4 5 are good ^^

The problem is that There are still No.4 would overlap with sim's withAsserts there.

Dolu1990, 2022-09-26:
But in some ways, i think we could just have withMasterAsserts and withSlaveAssumes, as Flow is still a master/slave interface (for instance)
That's fine if there is no withSlaveAsserts / withMasterAssumes

Readon, 2022-09-26:
> But in some ways, i think we could just have withMasterAsserts and withSlaveAssumes, as Flow is still a master/slave interface (for instance) That's fine if there is no withSlaveAsserts / withMasterAssumes

Good, then I will try to leave the existing functions there, and add withXXXAsserts and withXXXAssumes for consistency.
So that No. 4 remains only withCovers then.

Readon, 2022-09-26:
> Hmm, should the covers be different for master or slaves ? I mean covers is a bit more kind of a unified thing ? I'm not so much experienced in it.

Yes, the covers means that what would happen eventually. 
It describes the signal behavior of bus transaction, and can be applied for both master and slave ends.

Readon, 2022-09-26:
> withAsserts should be also fine for regular simulation too. Else has to be withFormalAsserts i think ? I think 2 3 4 5 are good ^^

The only withAsserts I found in existing code is in Formal Verification related code.

andreasWallner, 2022-09-27:
I'm sorry if this is a dumb question - not very experienced when it comes to formal verification...
How would this work for hierarchies? In the few examples I worked through I would replace `assume` in submodules with `assert` - would we do the same here for the whole hierarchy (optionally)?

I had taken the approach from the ZipCPU blog quite a while ago: https://zipcpu.com/formal/2018/12/18/skynet.html

Readon, 2022-09-27:
> I'm sorry if this is a dumb question - not very experienced when it comes to formal verification... How would this work for hierarchies? In the few examples I worked through I would replace `assume` [...]

Emmm, the problem is that formal verification would take too much CPU load and time when the design getting large.
The existing work in SpinalHDL is just concentrated on verifying facilities with basic function and logic. 

So hierarchical "leaf" replacing method that presented in ZipCPU are not applied now.
I don't think the method they apply is easy to fit in SpinalHDL. As I understand, what they do is removing the "leaf" instances from the design and put their assertion instead. 
It requires changing the DUT's structure, which makes the verification a little bit too complicated.

What the general roadmap, personally, is to verify basic facility enough, and then create enough helper functions for buses, which could help to decouple components in hierarchies.
Do you have any idea about this?

Dolu1990, 2022-09-29:
> I don't think the method they apply is easy to fit in SpinalHDL. As I understand, what they do is removing the "leaf" instances from the design and put their assertion instead.

hmm maybe it can be easy, go in the instance you want to remove the content, remove all sub component from the children list, remove all statments from the dslBody, and there you go, empty component, then you just need to do a myComponent.rework, and fill stuff you want in ^^

Would that works for the usage ?

Readon, 2022-09-29:
> hmm maybe it can be easy, go in the instance you want to remove the content, remove all sub component from the children list, remove all statments from the dslBody, and there you go, empty [...]

If so I need to try to build an example project first.
And the naming policy should be reconsidered.

Dolu1990, 2022-09-29:
```scala
object SubCompCleanerTest extends App {
  def flushComponent(c : Component): Unit ={
    c.children.clear()
    c.dslBody.foreachStatements(_.removeStatement())
  }

  class Toplevel extends Component{
    val a = Bool()
    val subInst = new Sub()
  }

  class Sub extends Component {
    val b = False
    val c = b ^ b
    val d = out(Rgb(5,6,5).getZero)
    val fifo = StreamFifo(Rgb(5,6,5), 64)
  }

  SpinalVerilog{
    val top = new Toplevel()
    flushComponent(top.subInst)
    top.subInst.rework(new AreaRoot {
      val miaou = False
      assert(miaou === False)
    })
    top
  }
}
```

Readon, 2022-10-01:
> ```scala
> object SubCompCleanerTest extends App {
>   def flushComponent(c : Component): Unit ={
>     c.children.clear()
>     c.dslBody.foreachStatements(_.removeStatement())
>   }
>
>   class Toplevel [...]

I think we should think about the hierarchy problem independently.
For example, if I defined the output signal as below
``` scala
  class Toplevel extends Component {
    val a = Bool()
    val subInst = new Sub()
    val dout = out(Rgb(5, 6, 5))
    dout := subInst.d
  }
```
Then subInst.d would be empty, if it is removed.
If we only add assume to the subInst.d then it still will report "no drive".

Furthermore, according to ZipCPU's article, when processing a hierarchy problem, what they do is just assert (no assumes) on the interfaces of the component to be reduced.
This contradicts with the SpinalHDL's "signal drive" check.

andreasWallner, 2022-10-02:
I think there might be a slight misunderstanding about what the conclusion of the ZipCPU blog was (and I'm also not saying it's the only way...).
They mention removal of the design code in the subcomponents as a possibility, but the option that they choose is to replace all interface assumptions with assertions and leave the code as-is. I understand that this can be quite costly w.r.t. runtime of the solver...

But: I'd agree that the details of the hierarchy could be worked out laters, I just wanted to mention it that we keep in mind that the generation of the properties will need to be hierarchical to some degree... That could actually reflect in the naming...

Readon, 2022-10-02:
> I think there might be a slight misunderstanding about what the conclusion of the ZipCPU blog was (and I'm also not saying it's the only way...). They mention removal of the design code in the [...]

If the design is still required in verification, then the cost has not been reduced.
On other hand, the assertion on all internal interfaces might make the cost even more, which can only helpful for debugging.

For a component, the assertion and assumption on interface with minimum assertions on internal logic (MAIL) can work also.
The rule 1 is targeting at MAIL of components, which can also helps to pass the "prove" though.

Dolu1990, 2022-10-03:
> Then subInst.d would be empty, if it is removed.
If we only add assume to the subInst.d then it still will report "no drive".

Hooo yes, i need to not remove the input / output, forgot about it. Here is the corrected example which preserve the component io : 

```scala
object SubCompCleanerTest extends App {
  def flushComponent(c : Component): Unit ={
    c.children.clear()
    c.dslBody.foreachStatements {
      case bt : BaseType if !bt.isDirectionLess =>
      case s => s.removeStatement()
    }
  }

  class Toplevel extends Component{
    val a = Bool()
    val subInst = new Sub()
    subInst.b := False
  }

  class Sub extends Component {
    val b = in Bool()
    val c = b ^ b
    val d = (Rgb(5,6,5).getZero)
    val fifo = StreamFifo(Rgb(5,6,5), 64)
  }

  SpinalVerilog{
    val top = new Toplevel()
    flushComponent(top.subInst)
    top.subInst.rework(new AreaRoot {
      val miaou = False
      assert(miaou === False)
    })
    top
  }
}
```

> Furthermore, according to ZipCPU's article, when processing a hierarchy problem, what they do is just assert (no assumes) on the interfaces of the component to be reduced.

Ahhhh so rulings things out ^^

Dolu1990, 2022-10-04:
Ahh, i mean, i'm assuming the user will provide the proper minimal hardware to have legal stuff in the submodule via rework.
mySubModule.rework{  mySubModule.myOutput := XXX  }

Readon, 2022-10-04:
The problem is that we have release 1.7.x with some interface using "withAsserts" now.
It seems that if we change the interface policy, it's hard for people to keep their code consistency.

Readon, 2022-10-04:
Perhaps it is better to keep "withAsserts" and "withAssumes" for formal verification.
And use withSimAsserts for simulation, because at least now there is no simulation code in the library that uses withAsserts.

Readon, 2022-10-04:
How about this:
1. all formal related functions return Area.
2. formalContext function can be used to create formal related logic in any cases.
3. there could be withAsserts, withAssumes and withCovers in formalContext returned Area.
4. for different cases, it could be.
A. For Component
    -. The minimum required asserts internally can be named as withAsserts or withFormalAsserts(need to modify existing interface).
B. For interfaces which implement IMasterSlave
     -. Name as withMasterAsserts, withMasterAssumes, withSlaveAsserts, withSlaveAssumes or withCovers.

So that add extra withXXXAsserts wrap function to Stream can keep the old code working and new consistent policy applies.

Dolu1990, 2022-10-04:
> Perhaps it is better to keep "withAsserts" and "withAssumes" for formal verification.

We can keep them for backward compatibilty, add withFormalAsserts which is just a proxy, and suggesting people to hae withFormalXxx  withSimXxx in the future ?

> In addition, it can be recommended to use rule 6, which requires the return type of these functions to be an Area, for better debugging.

Yes, I mean, and the best being a composite, allowing people to not store the reference while still having proper naming.

> 1. all formal related functions return Area.

return Composite is even better ^^, but area is good enough

> there could be withAsserts, withAssumes and withCovers in formalContext returned Area.

To be honnest i think it would be quite clearner to have withFormalXxx, it would be safer.

Else seems good to me.

Readon, 2022-10-05:
all right^^

closed 2022-10-06 by Readon