SpinalHDL/SpinalHDL #873: Feature: UIntToOh for faster simulation?

tanhongze, 2022-09-26:
Now, `UIntToOh` is implemented as
```
  def apply(value: UInt, width : Int): Bits = {
    val ret = Bits(width bits)
    for(i <- 0 until width){
      ret(i) := value === i
    }
    ret
  }
```
By using an alternative implement, we could make this simulation much faster(with no impact to synthesis):
```
  def apply(value: UInt, width : Int): Bits = {
    B(1, width bits) |<< value
  }
```

numero-744, 2022-09-26:
Can you open a PR, please? Else I can do it.

andreasWallner, 2022-09-26:
Out of curiosity, with which synthesis toolset did you compare the two?
This kind of reminds me of this blog entry: https://andy-knowles.github.io/one-hot-mux/ where Andy Knowles compared multiple HDL implementations of OH encoders.

tanhongze, 2022-09-26:
I just tested it under Vivado.
For a 5-32 decoder 10-1024 decoder, these implementations of `UIntToOh` would be mapped to exactly same synthesized design. (Thus, same net delay and same logic delay)

Dolu1990, 2022-09-27:
Merged

closed 2022-09-27 by Dolu1990