SpinalHDL/SpinalHDL #885: SpinalVerilog module name error

balanx, 2022-10-06:
Environment : JDK 18,  scala-2.12.16,   sbt-1.7.1,  spinalhdl 1.7.2

=== code begin ===
import spinal.core._
import spinal.lib._

//Hardware definition
class StreamArbiter_1 extends Component {
    val streamA, streamB, streamC = Stream(Bits(8 bits))
    val arbitredABC = StreamArbiterFactory.roundRobin.onArgs(streamA, streamB, streamC) 

    val streamD, streamE, streamF = Stream(Bits(8 bits))
    val arbitredDEF = StreamArbiterFactory.lowerFirst.noLock.onArgs(streamD, streamE, streamF)
}

object testApp extends  App {
    SpinalVerilog(new StreamArbiter_1)
}

=== code end ===

Result is "StreamArbiter_1_1.v"
But the expectation is "StreamArbiter_1.v"

Dolu1990, 2022-10-06:
Should be good in dev now ^^

Dolu1990, 2022-10-06:
Hoo, for code blocks, like 

```scala
val xxx = 42
```

it is 

\`\`\`scala
val xxx = 42
\`\`\`

commit 530a2001 (2022-10-06): Fix #885 Toplevel names are now allocated first