SpinalHDL/SpinalHDL #888: toByteArray function of BigInt works unexpected.

Readon, 2022-10-08:
I want to write the content of BigInt Array to a binary file by toByteArray function.
However, when the sign bit is 1, a negtive value, then BigInt works unexpected.

How about add a 'toBytes' function to BigInt as below to spinal.lib.sim package, so that some data like instruction can be operated with BigInt and write to file for hardware loading.
``` scala
object BigIntHelper {
  implicit class BigIntHelper(x: BigInt) {
    def toBytes(maxBits: Int = -1): Array[Byte] = {
      val len = maxBits.max(x.bitLength)
      val byteCount = (len + 7) / 8
      val occupyBytes = (x.bitLength + 7) / 8
      val out = Array.fill[Byte](byteCount) { 0 }
      for (i <- 0 until occupyBytes) {
        out(i) = ((x >> i * 8) & 0xff).toByte
      }
      out
    }
  }
}
```
Or is there any existing mechanism would help too?

kleinai, 2022-10-11:
I don't see any issue with negative numbers and `toByteArray`. Do you have a specific case?
```scala
object ByteTest {
  def printBigIntBytes(b: BigInt): String = "b" + b.toByteArray.map(
    _.toBinaryString.takeRight(8).reverse.padTo(8, "0").reverse.mkString
  ).mkString("_")

  def main(args: Array[String]): Unit = {
    for(i <- List(128, 127, 1, 0, -1, -128, -129)) {
      println(s"${i} -> ${printBigIntBytes(BigInt(i))}")
    }
  }
}
```
Yields:
```
128 -> b00000000_10000000
127 -> b01111111
1 -> b00000001
0 -> b00000000
-1 -> b11111111
-128 -> b10000000
-129 -> b11111111_01111111
```

Readon, 2022-10-11:
> I don't see any issue with negative numbers and `toByteArray`. Do you have a specific case?
>
> ```scala
> object ByteTest {
>   def printBigIntBytes(b: BigInt): String = "b" + b.toByteArray.map( [...]

I meet this problem long time ago, while simulation.
I just use your code to test it.
``` scala
object ByteTest {
  def printBigIntBytes(b: BigInt): String = {
    val bytes = b.toByteArray
    val out = bytes.length.toString + "_" + bytes
      .map(x => String.format("%02x", Byte.box(x)) // toBinaryString.takeRight(8).reverse.padTo(8, "0").reverse.mkString
      )
      .mkString("_")
    "b" + out
  }

  def main(args: Array[String]): Unit = {
    for (i <- List(128, 127, 1, 0, -1, -127, -128, -129, 0xfffffffe, 0xf1234567, 0xff234567, 0xffff4567)) {
      println(s"${i} -> ${printBigIntBytes(BigInt(i))}")
    }
  }
}
```
Yields:
```
128 -> b2_00_80
127 -> b1_7f
1 -> b1_01
0 -> b1_00
-1 -> b1_ff
-127 -> b1_81
-128 -> b1_80
-129 -> b2_ff_7f
-2 -> b1_fe
-249346713 -> b4_f1_23_45_67
-14465689 -> b4_ff_23_45_67
-47769 -> b3_ff_45_67
```
It can be seen that the length would be variable when a 64 bits data are received.

Dolu1990, 2022-10-12:
Sure, we can add such utilities.

Readon, 2022-10-12:
There is one problem that toByteArray use BigEndian as default, where high byte save in lower index.
If the order of bytes need to be consistent with toByteArray then code would be.
``` scala
  implicit class BigIntHelper(x: BigInt) {
    def toBytes(bits: Int = -1): Array[Byte] = {
      val raw = x.toByteArray
      val byteCount = if (bits < 0) raw.length else (bits + 7) / 8
      val out = Array.fill[Byte](byteCount) { 0 }
      for (i <- 0 until byteCount) {
        out(i) = ((x >> i * 8) & 0xff).toByte
      }
      out.reverse //here is the problem!
    }
  }
```

However, little endian is used more, as i think. 
At least the index should be consistent with byte order.
Here is the test code.
``` scala
  def printBigIntBytes(b: BigInt, maxBits: Int = -1): String = {
    // val bytes = b.toByteArray
    val bytes = b.toBytes(maxBits)
    val out = bytes.length.toString + "_" + bytes.zipWithIndex
      .map { case (x, id) =>
        // String.format(s"${id}: %02x", Byte.box(x))
        String.format("%02x", Byte.box(x))
      }
      .mkString("_")
    "b" + out
  }

  def main(args: Array[String]): Unit = {
    for (i <- List(128, 127, 1, 0, -1, -127, -128, -129, 0xfffffffe, 0xf1234567, 0xff234567, 0xffff4567)) {
      println(s"${i} -> ${printBigIntBytes(BigInt(i))}")
    }
    for (i <- List((BigInt(0xff) << 32) | BigInt(0xffff4567))) {
      println(s"${i} -> ${printBigIntBytes(i)}")
    }

    for (i <- List(128, 127, 1, 0, -1, -127, -128, -129, 0xfffffffe, 0xf1234567, 0xff234567, 0xffff4567)) {
      println(s"${i} -> ${printBigIntBytes(BigInt(i), 16)}")
    }
    for (i <- List((BigInt(0xff) << 32) | BigInt(0xffff4567))) {
      println(s"${i} -> ${printBigIntBytes(i, 16)}")
    }

    for (i <- List(128, 127, 1, 0, -1, -127, -128, -129, 0xfffffffe, 0xf1234567, 0xff234567, 0xffff4567)) {
      println(s"${i} -> ${printBigIntBytes(BigInt(i), 64)}")
    }
    for (i <- List((BigInt(0xff) << 32) | BigInt(0xffff4567))) {
      println(s"${i} -> ${printBigIntBytes(i, 64)}")
    }
  }
```

Dolu1990, 2022-10-13:
> However, little endian is used more, as i think.

Yes i agree. 
we could have toBytes being by default LITTLE   (Endianness type)

Readon, 2022-10-13:
> > However, little endian is used more, as i think.
> 
> Yes i agree. we could have toBytes being by default LITTLE (Endianness type)

Because actually there is no address here, just index, so endianness is not 100% fit.
I named the parameter byteOrderReverse to indicate the byte order of the cross-match.

Readon, 2022-10-13:
> What about doing such a thing right in the Spinal types?
> 
> * Avoids to do `toBigInt` to do `toBytes`, the latter could be used directly
> * The size would be the size of the signal.

ehhh, that would make a small patch large.:rofl:

numero-744, 2022-10-13:
What about something like this (not tested)?

```scala
implicit class SimBitVectorPimper(bt: BitVector) {

    def toInt    = getInt(bt)
    def toLong   = getLong(bt)
    def toBigInt = getBigInt(bt)
+   def toBytes: Array[Byte] = toBigInt.toBytes(bt.bitWidth)
+   def toBytesReverse: Array[Byte] = toBigInt.toBytes(bt.bitWidth, true)
    ...
```

Readon, 2022-10-14:
> What about something like this (not tested)?
> 
> ```scala
> implicit class SimBitVectorPimper(bt: BitVector) {
> 
>     def toInt    = getInt(bt)
>     def toLong   = getLong(bt)
>     def toBigInt = getBigInt(bt)
> +   def toBytes: Array[Byte] = toBigInt.toBytes(bt.getWidth)
> +   def toBytesReverse: Array[Byte] = toBigInt.toBytes(bt.getWidth, true)
>     ...
> ```

Good idea. However, I think the reverse is not that commonly used, so the definition can be remains to user.

PR #898 (2022-10-18): add toBytes to get fixed bytes from BigInt.
fix #888 .
closed 2022-10-18 by Readon