SpinalHDL/SpinalHDL #890: Assignment problem

balanx, 2022-10-10:
Environment : JDK 18, scala-2.12.16, sbt-1.7.1, spinalhdl 1.7.2

=== code begin ===
import spinal.core._
import spinal.lib._

case class Test() extends Component {
  var u = UInt(8 bits)
  u \= (3->false, default -> true)
}
=== code  end  ===

[error] D:\temp\test\test.scala:9:5: value \= is not a member of spinal.core.UInt
[error]   Expression does not convert to assignment because:
[error]     too many arguments (2) for method \: (that: spinal.core.UInt)spinal.core.UInt
[error]     expansion: Test.this.u = Test.this.u.\(3.$minus$greater(false), default.$minus$greater(true))
[error]   u \= (3->false, default -> true)
[error]     ^
[error] one error found
[error] (Compile / compileIncremental) Compilation failed

Readon, 2022-10-10:
use "u :=" instead. 
and "val u" is preferred, not "var u".

Dolu1990, 2022-10-11:
Also yes, i agree with Readon, in general, never use the \= operator, unless you want to do carry chaine/priority like stuff.

numero-744, 2022-10-14:
@balanx is it okay for you?

balanx, 2022-10-15:
OK.
Thanks Readon. Thanks Dolu. Thanks numero !

closed 2022-10-15 by Readon