SpinalHDL/SpinalHDL #893: Axi4StreamWidthAdapter works incorrect in some cases

xueweiwujxw, 2022-10-12:
Here is my example:
```scala
package Enc

import spinal.core._
import spinal.lib._

import spinal.sim._
import spinal.core.sim._
import spinal.lib.bus.amba4.axis._

import spinal.core.fiber.Handle

case class streamBitsConvert(
    inaxisConfig: Axi4StreamConfig,
    outaxisConfig: Axi4StreamConfig
) extends Component {
  val io = new Bundle {
    val ins = slave(Axi4Stream(inaxisConfig))
    val outs = master(Axi4Stream(outaxisConfig))
  }
  val a = Axi4StreamWidthAdapter(io.ins, io.outs)
}

object streamBitsConverGen extends App {
  import spinal.lib.sim.StreamReadyRandomizer
  SimConfig.withFstWave
    .doSim(
      streamBitsConvert(
        Axi4StreamConfig(
          dataWidth = 4,
          useKeep = true,
          useLast = true
        ),
        Axi4StreamConfig(
          dataWidth = 1,
          useKeep = true,
          useLast = true
        )
      )
    ) { dut =>
      dut.clockDomain.forkStimulus(10)
      dut.io.ins.payload.keep #= (scala.math.pow(
        2,
        dut.inaxisConfig.dataWidth
      ) - 1).toInt
      dut.io.ins.payload.last #= false
      // StreamReadyRandomizer(dut.io.outs, dut.clockDomain)
      dut.io.outs.ready #= true
      for (idx <- 0 until 100) {
        if (idx > 10) {
          dut.io.ins.valid #= true
          dut.io.ins.payload.data #= idx
        } else {
          dut.io.ins.valid #= false
        }
        dut.clockDomain.waitSampling()
      }
    }
}
```
The input ready signal was pull down when Axi4StreamWidthAdaptor had received 3 clocks, and it would be never rise.
[image]
However when I change the input dataWidth to 64, it seems to be right.
[image]
Scala version: 2.12.15
Spinal vesion: 1.7.2

Dolu1990, 2022-10-12:
@kleinai :)

xueweiwujxw, 2022-10-12:
There is another problem.
```scala
package Enc

import spinal.core._
import spinal.lib._
import spinal.lib.bus.amba4.axis._

import spinal.sim._
import spinal.core.sim._

import spinal.core.fiber.Handle
import lib._

/**
 * axis width adaptor with packet mode
 *
 * @param packetMode
 * @param inAxisConf
 * @param outAxisConf
 * @param FIFODepth
 */
case class streamWidthConverter(
    inAxisConf: Axi4StreamConfig,
    outAxisConf: Axi4StreamConfig,
    FIFODepth: Int = 1024
) extends Component {
  val io = new Bundle {
    val ins = slave(Axi4Stream(inAxisConf))
    val outs = master(Axi4Stream(outAxisConf))
  }
  Axi4StreamWidthAdapter(io.ins, io.outs)
}

object streamWidthConverterGen extends App {
  SpinalConfig().generateVerilog(
    streamWidthConverter(
      Axi4StreamConfig(dataWidth = 8, useLast = true, useKeep = true),
      Axi4StreamConfig(dataWidth = 4, useLast = true, useKeep = true)
    )
  )
}

object streamWidthConverterSim extends App {
  def WriteCtlStream(
      s_stream: Axi4Stream.Axi4Stream,
      clock: Handle[ClockDomain],
      num: Int
  ) {
    var numIn = num
    s_stream.valid #= true
    s_stream.last #= false
    s_stream.data #= 0x1ff0
    s_stream.keep #= 0xf
    clock.waitSampling(1)
    while (!s_stream.ready.toBoolean) {
      clock.waitSampling(1)
    }
    s_stream.data #= num
    clock.waitSampling(1)
    while (!s_stream.ready.toBoolean) {
      clock.waitSampling(1)
    }

    while (numIn > 0) {
      if (numIn <= 4) {
        s_stream.last #= true
      } else
        s_stream.last #= false
      var numMut = numIn - 1
      if (numMut > 127) {
        s_stream.keep #= 0xf
        s_stream.data #= ((numMut % 128) << 24 | ((numMut - 1) % 128) << 16 | ((numMut - 2) % 128) << 8 | ((numMut - 3) % 128))
        numIn = numIn - 4
      } else {
        if (numMut > 2) {
          s_stream.data #= (numMut << 24 | (numMut - 1) << 16 | (numMut - 2) << 8 | (numMut - 3))
          s_stream.keep #= 0xf
          numIn = numIn - 4
        } else {
          numMut match {
            case 2 =>
              s_stream.data #= (numMut << 16 | (numMut - 1) << 8 | (numMut - 2))
              s_stream.keep #= 0x7
              numIn = numIn - 3
            case 1 =>
              s_stream.data #= (numMut << 8 | (numMut - 1))
              s_stream.keep #= 0x3
              numIn = numIn - 2
            case 0 =>
              s_stream.data #= (numMut)
              s_stream.keep #= 0x1
              numIn = numIn - 1
          }
        }
      }
      clock.waitSampling()
      while (!s_stream.ready.toBoolean) {
        clock.waitSampling()
      }
    }

    s_stream.valid #= false
    s_stream.last #= false
    clock.waitSampling(1)
  }
  SimConfig.withFstWave.doSim(
    streamWidthConverter(
      Axi4StreamConfig(dataWidth = 4, useLast = true, useKeep = true),
      Axi4StreamConfig(dataWidth = 8, useLast = true, useKeep = true)
    )
  ) { dut =>
    dut.clockDomain.forkStimulus(10)
    dut.io.ins.flattenForeach(x => if (x.isInput) x.assignBigInt(0))
    dut.io.outs.ready #= true

    dut.clockDomain.waitSampling(20)

    WriteCtlStream(dut.io.ins, dut.clockDomain, 66)

    WriteCtlStream(dut.io.ins, dut.clockDomain, 64)

    WriteCtlStream(dut.io.ins, dut.clockDomain, 65)

    WriteCtlStream(dut.io.ins, dut.clockDomain, 67)

    WriteCtlStream(dut.io.ins, dut.clockDomain, 68)

    dut.clockDomain.waitSampling(2000)
  }
}
```
When I try to convert a 32-bit wide stream of 64 bytes to a 64-bit wide stream, the last signal is delayed one clock.
[image]
I think this problem only occurs when the number of bytes of the input stream is a multiple of the output bit width

kleinai, 2022-10-12:
I'm still unable to reproduce your first bug. Could you take some screenshots of the internal state? Maybe I can derive a little of what's going on. I'm on your version of Spinal and Scala. Try updating your cache?

The second bug actually isn't a bug. It's within spec that TLAST may come on a beat with all null or positional bytes (aka no data).
I can reproduce this "bug" so I can try to make it behave in a way that's a little more user friendly. See section 2.5.1 of the specification (ARM IHI 0051A).

xueweiwujxw, 2022-10-13:
> I'm still unable to reproduce your first bug. Could you take some screenshots of the internal state? Maybe I can derive a little of what's going on. I'm on your version of Spinal and Scala. Try [...]

For the first bug, I cleared the cache and rerun it. This is the fst file. 
[test.tar.gz](https://github.com/SpinalHDL/SpinalHDL/files/9769677/test.tar.gz) or [google drive](https://drive.google.com/file/d/1wzCpYtBf835EbLGTrhslqVoyJ2X4Qnhw/view)
For the second "bug", I have always thought that the last data is also valid, so the last data will also be processed in other modules with TKEEP. But I did not consider this case that the TKEEP is 0 with TLAST asserted. I will try to handle this case.
By the way, is there anything that can make it behave in the way I thought?

kleinai, 2022-10-14:
Pushed a fix for the delayed last beat. Fill checker was off-by-one.

kleinai, 2022-10-14:
Something is really weird with your simulation.

Here's the state of my write buffer for the first few beats:
[image]

Here's your first few beats:
[image]

Buffer valid shouldn't have any holes in data. It represents valid bytes in the buffer regardless of the state of TKEEP on those bytes. On the first beat that buffer should be 0x0F which is 4 bytes of data. Yours shows data 6 bytes (with a hole) which should not happen.

I recommend pulling down the new TLAST update and trying again.

xueweiwujxw, 2022-10-14:
Maybe the verilator is the root cause. It seems to be right after I upgraded it from v4.040 to v4.228.

closed 2022-10-16 by xueweiwujxw