SpinalHDL/SpinalHDL #894: [Bug] setWhen incorrect

dokleina, 2022-10-12:
I believe `setWhen` isn't working as it should.

Below is a snippet of what I'm using it for:
``` scala
...
val running = Reg(Bool()) init(False)
running.setWhen(io.start)
when(running) {
    // This should be done on the cycle after "io.start"
    ...
}
...
```
\
I'd expect the `setWhen` to set the `running` flag for the next cycle when it sees the `io.start`. However, looking at a sim waveform I see `running` getting set on the same cycle.

Instead if I do this:
``` scala
...
val running = Reg(Bool()) init(False)
when(io.start) {
    running := True
}
```
The waveform looks as expected: `running` is set on cycle after `io.start`.
\
I have noticed that `setWhen` works when used in a chain with the register declaration:
``` scala
val running = (Reg(Bool()) init(False)).setWhen(io.start)
```
\
\
Am I wrong in my understanding of `setWhen`? Is this a bug or intended?

Dolu1990, 2022-10-12:
Hi,

So running.setWhen(io.start) whould be strictly equivalent to  when(io.start){ running := True }
I never had issues as described.

Do you have a small self contained example (including simulation testbench) for us to reproduce ?
Or check the generated verilog  ?

closed 2022-10-12 by dokleina

Dolu1990, 2022-10-12:
No worries :)