SpinalHDL/SpinalHDL #895: "Last valid assignment wins" inconsistent

balanx, 2022-10-13:
e.g. 1 passed.
```scala
val x, y = Bool()           // Define two combinational signals
val result = UInt(8 bits)   // Define a combinational signal

result := 1
when(x) {
  result := 2
  when(y) {
    result := 3
  }
}
```
but e.g. 2 failed
```scala
class Test extends Component {
  val x, y, result = UInt(8 bits)
  result := ~x          // step-1
  result :=  result + 1 // step-2
  result :=  result + y // step-3
}
```
why ? is it because there is difference at "when" statement  ?

e.g. 2 is only a small example. in some algorithm design case, some calculating steps may be needed, like step-1, step-2, ...
so I hope spinal can support statements just like  e.g. 2.

numero-744, 2022-10-13:
Hello,

Let's play a little bit with assignments :wink:

## Example 1

<details><summary>Click me to show code with test</summary>

```scala
object Example1 extends App {
  import spinal.core._

  class Example1 extends Component {
    val x, y = in Bool ()
    val result = out UInt (8 bits)

    result := 1
    when(x) {
      result := 2
      when(y) {
        result := 3
      }
    }
  }

  sim.SimConfig.doSim(new Example1) { dut =>
    import sim._

    dut.x #= false
    dut.y #= false
    sleep(1)
    assert(dut.result.toInt == 1)

    dut.x #= false
    dut.y #= true
    sleep(1)
    assert(dut.result.toInt == 1)

    dut.x #= true
    dut.y #= false
    sleep(1)
    assert(dut.result.toInt == 2)

    dut.x #= true
    dut.y #= true
    sleep(1)
    assert(dut.result.toInt == 3)
  }
}
```

</details>

It works as expected:`when` is used to build muxes. The schematic is:

```
          _
    _    | \
   | \ 1-|F|
 2-|F|   | |--> result
   | |---|T|
 3-|T|   |_/
   |_/    |
    | x >-·
y >-·
```

There are no assignment overlaps because of `when` which brings a context with a condition: assignment above the `when` is the default case, when the condition is false.

If you want to play more (just copy-paste the code above and edit it, it can be run as-is as it extends `App`):
[... 221 lines cut ...]

</details>

## The docs

There is an issue in the docs (I'm gonna PR it right now): https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Semantic/assignments.html#id1

```scala
object DocsExampleIsWrong extends App {
  import spinal.core._

  class DocsExampleIsWrong extends Component {
    val a, b, c = UInt(4 bits)
    //a.allowOverride // this makes the example work
    a := 0
    b := a
    a := 1 // ASSIGNMENT OVERLAP
    c := a
  }

  SpinalConfig().generateVerilog(new DocsExampleIsWrong)
}
```

I think it is because this part has been written when assignment overlap was not checked, it was considered correct using the rule of last assignment (correct as far as it does not implies Xes in simulation). Then is has been found that it is often a mistake to write such code ("Why do I have a 1 in a? I wrote `a := 0` in my code and have no muxex!") so a check has been added, and the doc has not been updated since.

balanx, 2022-10-14:
It's very kind of you to explain it so in detail and I was persuaded. Thanks u so much.

closed 2022-10-14 by numero-744