SpinalHDL/SpinalHDL #896: Preserving names problem

balanx, 2022-10-13:
spinal code
···scala
class Test extends Component {

  // Use the current clock domain : 100MHz
  val areaStd = new Area {
    val counter = out(CounterFreeRun(16).value)
  }

  // Slow the current clockDomain by 4 : 25 MHz
  val areaDiv4 = new SlowArea(4) {
    val counter = out(CounterFreeRun(16).value)
  }

  // Slow the current clockDomain to 50MHz
  val area50Mhz = new SlowArea(50 MHz) {
    val counter = out(CounterFreeRun(16).value)
  }
}
···
output verilog
···verilog
// Generator : SpinalHDL v1.7.2    git head : 08fc866bebdc40c471ebe327bface63e34406489
// Component : Test

`timescale 1ns/1ps

module Test (
  output reg [3:0]    _zz_3,
  output reg [3:0]    _zz_6,
  output reg [3:0]    _zz_9,
  input               clk,
  input               reset
);

  wire       [3:0]    _zz__zz_2;
  wire       [0:0]    _zz__zz_2_1;
  wire       [3:0]    _zz__zz_5;
  wire       [0:0]    _zz__zz_5_1;
  wire       [3:0]    _zz__zz_8;
  wire       [0:0]    _zz__zz_8_1;
  wire                _zz_1;
  reg        [3:0]    _zz_2;
  reg        [1:0]    _zz_when_ClockDomain_l369;
  wire                when_ClockDomain_l369;
  reg                 when_ClockDomain_l369_regNext;
  wire                _zz_4;
  reg        [3:0]    _zz_5;
  reg        [0:0]    _zz_when_ClockDomain_l369_1;
  wire                when_ClockDomain_l369_1;
  reg                 when_ClockDomain_l369_1_regNext;
  wire                _zz_7;
  reg        [3:0]    _zz_8;

  assign _zz__zz_2_1 = _zz_1;
  assign _zz__zz_2 = {3'd0, _zz__zz_2_1};
  assign _zz__zz_5_1 = _zz_4;
  assign _zz__zz_5 = {3'd0, _zz__zz_5_1};
  assign _zz__zz_8_1 = _zz_7;
  assign _zz__zz_8 = {3'd0, _zz__zz_8_1};
  always @(*) begin
    _zz_2 = (_zz_3 + _zz__zz_2);
    if(1'b0) begin
      _zz_2 = 4'b0000;
    end
  end

  assign _zz_1 = 1'b1;
  assign when_ClockDomain_l369 = (_zz_when_ClockDomain_l369 == 2'b11);
  always @(*) begin
    _zz_5 = (_zz_6 + _zz__zz_5);
    if(1'b0) begin
      _zz_5 = 4'b0000;
    end
  end

  assign _zz_4 = 1'b1;
  assign when_ClockDomain_l369_1 = (_zz_when_ClockDomain_l369_1 == 1'b1);
  always @(*) begin
    _zz_8 = (_zz_9 + _zz__zz_8);
    if(1'b0) begin
      _zz_8 = 4'b0000;
    end
  end

  assign _zz_7 = 1'b1;
  always @(posedge clk or posedge reset) begin
    if(reset) begin
      _zz_3 <= 4'b0000;
      _zz_when_ClockDomain_l369 <= 2'b00;
      when_ClockDomain_l369_regNext <= 1'b0;
      _zz_when_ClockDomain_l369_1 <= 1'b0;
      when_ClockDomain_l369_1_regNext <= 1'b0;
    end else begin
      _zz_3 <= _zz_2;
      _zz_when_ClockDomain_l369 <= (_zz_when_ClockDomain_l369 + 2'b01);
      if(when_ClockDomain_l369) begin
        _zz_when_ClockDomain_l369 <= 2'b00;
      end
      when_ClockDomain_l369_regNext <= when_ClockDomain_l369;
      _zz_when_ClockDomain_l369_1 <= (_zz_when_ClockDomain_l369_1 + 1'b1);
      if(when_ClockDomain_l369_1) begin
        _zz_when_ClockDomain_l369_1 <= 1'b0;
      end
      when_ClockDomain_l369_1_regNext <= when_ClockDomain_l369_1;
    end
  end

  always @(posedge clk or posedge reset) begin
    if(reset) begin
      _zz_6 <= 4'b0000;
    end else begin
      if(when_ClockDomain_l369_regNext) begin
        _zz_6 <= _zz_5;
      end
    end
  end

  always @(posedge clk or posedge reset) begin
    if(reset) begin
      _zz_9 <= 4'b0000;
    end else begin
      if(when_ClockDomain_l369_1_regNext) begin
        _zz_9 <= _zz_8;
      end
    end
  end

endmodule
···

There is no signals or ports named “areaStd_counter, areaDiv4_counter, area50Mhz_counter” !

numero-744, 2022-10-13:
What you are doing is aliasing signals defined in CounterFreeRun(16), that you use without giving it a name so it also has dummy names.

Note about aliasing:

```scala
val s0 = Bool // creating a new not-yet-assigned signal
val s1 = s0 // aliasing s0
s1 := True // s0 is True too because s1 is an alias for s0
val s2 = a | b // creating a new assigned signal from an expression
when(cond) { s2 := s1 } // only affects s2
```

What you want is maybe:

```scala

object AreaExample extends App {
  import spinal.core._
  import spinal.lib.CounterFreeRun

  class AreaExample extends Component {
    // Use the current clock domain: 100MHz
    val areaStd = new Area {
      val counter = out UInt ()
      val c = CounterFreeRun(16)
      counter := c.value
    }

    // Slow the current clockDomain by 4 : 25 MHz
    val areaDiv4 = new SlowArea(4) {
      val counter = out UInt ()
      val c = CounterFreeRun(16)
      counter := c.value
    }

    // Slow the current clockDomain to 50MHz
    val area50Mhz = new SlowArea(50 MHz) {
      val counter = out UInt ()
      val c = CounterFreeRun(16)
      counter := c.value
    }
  }

  SpinalConfig(defaultClockDomainFrequency = FixedFrequency(100 MHz))
    .generateVerilog(new AreaExample)
}
```

```verilog

module AreaExample (
  output     [3:0]    areaStd_counter,
  output     [3:0]    areaDiv4_counter,
  output     [3:0]    area50Mhz_counter,
  input               clk,
  input               reset
);

  wire       [3:0]    _zz_areaStd_c_valueNext;
  wire       [0:0]    _zz_areaStd_c_valueNext_1;
  wire       [3:0]    _zz_areaDiv4_c_valueNext;
  wire       [0:0]    _zz_areaDiv4_c_valueNext_1;
  wire       [3:0]    _zz_area50Mhz_c_valueNext;
  wire       [0:0]    _zz_area50Mhz_c_valueNext_1;
  wire                areaStd_c_willIncrement;
  wire                areaStd_c_willClear;
  reg        [3:0]    areaStd_c_valueNext;
  reg        [3:0]    areaStd_c_value;
  wire                areaStd_c_willOverflowIfInc;
  wire                areaStd_c_willOverflow;
[... 82 lines cut ...]
    end
  end

  always @(posedge clk or posedge reset) begin
    if(reset) begin
      areaDiv4_c_value <= 4'b0000;
    end else begin
      if(when_ClockDomain_l369_regNext) begin
        areaDiv4_c_value <= areaDiv4_c_valueNext;
      end
    end
  end

  always @(posedge clk or posedge reset) begin
    if(reset) begin
      area50Mhz_c_value <= 4'b0000;
    end else begin
      if(when_ClockDomain_l369_1_regNext) begin
        area50Mhz_c_value <= area50Mhz_c_valueNext;
      end
    end
  end

endmodule
```

Dolu1990, 2022-10-13:
Ahhh  val xxx = CounterFreeRun(16).value

is a special case : 
- The "value" is named by CounterFreeRun, because it is first part of the CounterFreeRun Area
- "xxx" is proposed as name for the value signal, but is refused, as the value signal is part of the CounterFreeRun Area
- the CounterFreeRun area is never itself named.

So, yeah, you realy need to capture CounterFreeRun area in a val to provide it a name.

One workaround for your case is : 
val x = CounterFreeRun(16).value.unsetName()

balanx, 2022-10-14:
Got it !
Thanks numero !
Thanks Dolu !

closed 2022-10-14 by numero-744