SpinalHDL/SpinalHDL #897: allowOverride misses braces

numero-744, 2022-10-13:
In [core/src/main/scala/spinal/core/Data.scala:454](https://github.com/SpinalHDL/SpinalHDL/blob/6edcb78476048abc3d134bc79351bce98298f983/core/src/main/scala/spinal/core/Data.scala#L454)

```scala
  def allowOverride: this.type = {
    addTag(allowAssignmentOverride)
  }
```

It seems this method (and the subsequent methods) modifies `this` via `addTag`. As it is not pure, Scala convention / recommendation is to add braces to it, both when declaring and calling.

The current declaration, without braces, prevents using braces to call the method, so it prevents applying recommendations, which is an issue. Adding braces in the declaration would make it possible to add them (or omit them in at least most of the cases).

Dolu1990, 2022-10-13:
> The current declaration, without braces, prevents using braces to call the method

Sorry, i'm not sure to understand, do you have some code to show what is prevented ?

numero-744, 2022-10-13:
```scala
object DocsExampleIsWrong extends App {
  import spinal.core._

  class DocsExampleIsWrong extends Component {
    val a, b, c = UInt(4 bits)
    a.allowOverride() // build error
    a := 0
    b := a
    a := 1 // ASSIGNMENT OVERLAP
    c := a
  }

  SpinalConfig().generateVerilog(new DocsExampleIsWrong)
}
```

In Scala it is recommended to put braces when calling `allowOverride` as it is modifying a, but here it fails because of the braces after `allowOverride`, as it is defined without braces. Actually, as it returns the current signal, braces are applied on the current signal so it expects something strange like (which builds and can be run):

```scala
val d = a.allowOverride(2 downto 0)
// same as
a.allowOverride
val d = a(2 downto 0)
```

Dolu1990, 2022-10-17:
Ahh so, here it is kind of conflicting interreset between avoiding verbosity on function which can be used in a builder pattern way : 
```scala
val xxx = UInt(8 bits) allowOverride allowXxx randInit  and so on
```

You can see those kind of stuff more like tags.

But overall def allowOverride()   also allow to call the function without the braces, so we can update def allowOverride with braces, that's fine for me.
Do you feel good about doing a pr ^^?

numero-744, 2022-10-17:
I'll do a PR, we will see if it breaks code :D

closed 2022-10-19 by numero-744