SpinalHDL/SpinalHDL #912: error of addAttribute(new AttributeFlag("name", COMMENT_ATTRIBUTE))

loonggi, 2022-10-19:
I want to generate a 4G Byte sparese Memory for simulation in VCS. The code is as below,
```
val memory = Mem(Bits(32 bits), 1 GiB)
memory.addAttribute(new AttributeFlag("sparse", COMMENT_ATTRIBUTE))
```
However, the generation of Verilog is `reg [31:0] memory [0:1073741823] /*sparse*/;` not  `reg /*sparse*/ [31:0] memory [0:1073741823];`, which not meet the syntax requirement for sparse memory models.

Thanks!

Dolu1990, 2022-10-20:
Hi,

Ahh it seems it kind of depend the tool, expectation changes.

VCS does not support (* xxx *) reg .....   like attributes ?

If not, we can add a COMMENT_EARLY_ATTRIBUTE tag which would generate things as it want.

loonggi, 2022-10-21:
Hi, thanks for you reply.
As far as I know, VCS does not support (* xxx *) reg ..... like attributes, eg., neither for sparse memory model nor for four states reg for simulation likes `reg /*4value*/ r1;`.

Dolu1990, 2022-10-21:
I pushed in dev a feature : 

```scala
    val memory = Mem(Bits(32 bits), 1 GiB)
    memory.addAttribute(new AttributeFlag("sparse", COMMENT_EARLY_ATTRIBUTE))
```

Should meet your needs. Any idea for a better naming to COMMENT_EARLY_ATTRIBUTE is welcome

numero-744, 2022-10-21:
What do you think about `COMMENT_TYPE_ATTRIBUTE`, because it is just after the integer data type `reg`?

(I was not sure `reg` was defined as a type, I have found it in *SystemVerilog 3.1a Language Reference Manual*, Section 3.3 Integer data types, Table 3-1: Integer data types)

loonggi, 2022-10-22:
`/* name */` is a pragma or meta commet for `reg` as depicted in  VCS User Guide, 3.Modeling Your Design, Using Sparse Memory Models.

closed 2022-10-22 by loonggi

Dolu1990, 2022-10-22:
> COMMENT_TYPE_ATTRIBUTE

Sound good to me