SpinalHDL/SpinalHDL #922: sbt compile failed

balanx, 2022-10-22:
I downloaded the latest sources, and got the following error information.

```
D:\msys64\home\Administrator\SpinalHDL-dev>sbt
[info] welcome to sbt 1.7.1 (Oracle Corporation Java 18)
[info] loading settings for project spinalhdl-dev-build from plugin.sbt ...
[info] loading project definition from D:\msys64\home\Administrator\SpinalHDL-dev\project
[warn]
[warn]  Note: Some unresolved dependencies have extra attributes.  Check that these dependencies exist with the requested attributes.
[warn]          com.jsuereth:sbt-pgp:1.1.0 (sbtVersion=1.0, scalaVersion=2.12)
[warn]          com.eed3si9n:sbt-unidoc:0.4.1 (sbtVersion=1.0, scalaVersion=2.12)
[warn]          com.eed3si9n:sbt-assembly:0.14.10 (sbtVersion=1.0, scalaVersion=2.12)
[warn]
[warn]  Note: Unresolved dependencies path:
[error] sbt.librarymanagement.ResolveException: Error downloading com.jsuereth:sbt-pgp;sbtVersion=1.0;scalaVersion=2.12:1.1.0
[error]   Not found
[error]   Not found
[error]   not found: D:\.sbt\.ivy2\localcom.jsuereth\sbt-pgp\scala_2.12\sbt_1.0\1.1.0\ivys\ivy.xml
[error]   not found: https://repo.huaweicloud.com/repository/maven/com/jsuereth/sbt-pgp_2.12_1.0/1.1.0/sbt-pgp-1.1.0.po
[error]   not found: https://repo1.maven.org/maven2/com/jsuereth/sbt-pgp_2.12_1.0/1.1.0/sbt-pgp-1.1.0.pom
[error]   not found: https://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/com.jsuereth/sbt-pgp/scala_2.12/sbt_1.0/1.1.0/ivys/artifact.xml
[error]   not found: https://repo.scala-sbt.org/scalasbt/maven-releases/com/jsuereth/sbt-pgp_2.12_1.0/1.1.0/sbt-pgp-1.1.0.pom
[error]   not found: https://repo.scala-sbt.org/scalasbt/maven-snapshots/com/jsuereth/sbt-pgp_2.12_1.0/1.1.0/sbt-pgp-1.1.0.pom
[error]   not found: https://repo.typesafe.com/typesafe/ivy-releases/com.jsuereth/sbt-pgp/1.1.0/ivys/ivy.xml
[error]   not found: https://repo.scala-sbt.org/scalasbt/ivy-snapshots/com.jsuereth/sbt-pgp/1.1.0/ivys/ivy.xml
[error] Error downloading com.eed3si9n:sbt-unidoc;sbtVersion=1.0;scalaVersion=2.12:0.4.1
[error]   Not found
[error]   Not found
```

Readon, 2022-10-22:
I just tried on my existing environment, there is no problem.

Someone report that "sbt-pgp" package has been renamed recently. 
I am not familiar with the distribution policy of sbt and java world, but that might be a problem for new user without existing working environment.

numero-744, 2022-10-22:
```
% sbt
[info] welcome to sbt 1.6.0 (N/A Java 17.0.4.1)
```

I wonder if it can be linked to `sbt` version. @Readon what version of sbt do you have?

Readon, 2022-10-22:
1.6.0 on Ubuntu 20 & 1.5.5 on Windows
Both works well.

Readon, 2022-10-22:
It had to be tried in a new, clean environment, but I didn't have one on hand.

numero-744, 2022-10-22:
Do you think the issue is can be that 1.7.1 is too recent?

I'll try to dockerize a new clean environment to check with 1.7.1.

Readon, 2022-10-22:
On my Windows, there are only pacakge files on "...\Coursier\cache\v1\https\repo.scala-sbt.org\scalasbt\sbt-plugin-releases\com.jsuereth\sbt-pgp\scala_2.12\sbt_1.0\1.1.0"
I think use coursier to install the sbt might helps.

balanx, 2022-10-22:
I changed to sbt-1.6.0 and  there is still the same error information.

Readon, 2022-10-22:
> I changed to sbt-1.6.0 and there is still the same error information.

Would you please try to install sbt with coursier first?

balanx, 2022-10-22:
Sorry, I am not familiar with  coursier. May you give me some guide information more ?

Readon, 2022-10-22:
This [link](https://get-coursier.io/docs/sbt-coursier) might helps.

Dolu1990, 2022-10-24:
You may had some bad-luck where some online server were offline ?
Is the issue still perssisting ?

Dolu1990, 2022-10-24:
I updated the sbt-pgp depedency.
@balanx It may be fine now ?

balanx, 2022-10-24:
There is another error information.
```
PS D:\msys64\home\WIN10\SpinalHDL-dev> sbt
[info] welcome to sbt 1.6.2 (Oracle Corporation Java 18)
[info] loading settings for project spinalhdl-dev-build from plugin.sbt ...
[info] loading project definition from D:\msys64\home\WIN10\SpinalHDL-dev\project
[warn]
[warn]  Note: Some unresolved dependencies have extra attributes.  Check that these dependencies exist with the requested attributes.
[warn]          com.eed3si9n:sbt-unidoc:0.4.1 (sbtVersion=1.0, scalaVersion=2.12)
[warn]          com.eed3si9n:sbt-assembly:0.14.10 (sbtVersion=1.0, scalaVersion=2.12)
[warn]
[warn]  Note: Unresolved dependencies path:
[error] sbt.librarymanagement.ResolveException: Error downloading com.eed3si9n:sbt-unidoc;sbtVersion=1.0;scalaVersion=2.12:0.4.1
[error]   Not found
[error]   Not found
[error]   not found: D:\.sbt\.ivy2\localcom.eed3si9n\sbt-unidoc\scala_2.12\sbt_1.0\0.4.1\ivys\ivy.xml
[error]   not found: https://repo.huaweicloud.com/repository/maven/com/eed3si9n/sbt-unidoc_2.12_1.0/0.4.1/sbt-unidoc-0.4.1.pom
[error]   not found: https://maven.aliyun.com/repository/public/com/eed3si9n/sbt-unidoc_2.12_1.0/0.4.1/sbt-unidoc-0.4.1.pom
[error]   not found: https://maven.aliyun.com/repository/central/com/eed3si9n/sbt-unidoc_2.12_1.0/0.4.1/sbt-unidoc-0.4.1.pom
[error]   not found: https://repo1.maven.org/maven2/com/eed3si9n/sbt-unidoc_2.12_1.0/0.4.1/sbt-unidoc-0.4.1.pom
[error]   not found: https://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/com.eed3si9n/sbt-unidoc/scala_2.12/sbt_1.0/0.4.1/ivys/artifact.xml
[error]   not found: https://repo.scala-sbt.org/scalasbt/maven-releases/com/eed3si9n/sbt-unidoc_2.12_1.0/0.4.1/sbt-unidoc-0.4.1.pom
[error]   not found: https://repo.scala-sbt.org/scalasbt/maven-snapshots/com/eed3si9n/sbt-unidoc_2.12_1.0/0.4.1/sbt-unidoc-0.4.1.pom
```

Dolu1990, 2022-10-24:
@balanx Ahhh, i just pushed another update for  sbt-unidoc, it may be fine now, if not, let's me know :)

balanx, 2022-10-24:
Ahhh :)
com.eed3si9n:sbt-assembly:2.0.0

```
[info] welcome to sbt 1.6.2 (Oracle Corporation Java 18)
[info] loading settings for project spinalhdl-dev-build from plugin.sbt ...
[info] loading project definition from D:\msys64\home\WIN10\SpinalHDL-dev\project
[warn]
[warn]  Note: Some unresolved dependencies have extra attributes.  Check that these dependencies exist with the requested attributes.
[warn]          com.eed3si9n:sbt-assembly:0.14.10 (sbtVersion=1.0, scalaVersion=2.12)
[warn]
[warn]  Note: Unresolved dependencies path:
[error] sbt.librarymanagement.ResolveException: Error downloading com.eed3si9n:sbt-assembly;sbtVersion=1.0;scalaVersion=2.12:0.14.10
[error]   Not found
[error]   Not found
[error]   not found: D:\.sbt\.ivy2\localcom.eed3si9n\sbt-assembly\scala_2.12\sbt_1.0\0.14.10\ivys\ivy.xml
[error]   not found: https://repo.huaweicloud.com/repository/maven/com/eed3si9n/sbt-assembly_2.12_1.0/0.14.10/sbt-assembly-0.14.10.pom
[error]   not found: https://maven.aliyun.com/repository/public/com/eed3si9n/sbt-assembly_2.12_1.0/0.14.10/sbt-assembly-0.14.10.pom
[error]   not found: https://maven.aliyun.com/repository/central/com/eed3si9n/sbt-assembly_2.12_1.0/0.14.10/sbt-assembly-0.14.10.pom
```

balanx, 2022-10-25:
com.eed3si9n:sbt-assembly:1.1.0
There is NO error after modifying it like above

Dolu1990, 2022-10-25:
@numero-744 @balanx Thanks :D

Dolu1990, 2022-10-25:
Merged

closed 2022-10-25 by Dolu1990