SpinalHDL/SpinalHDL #923: breaking changes in sim/Agent api

numero-744, 2022-10-22:
```diff
-onWriteByte(address: BigInt, data: Byte): Unit
+onWriteByte(address: BigInt, data: Byte, id: Int): Unit
-def onLast(id: Int): Unit
```

Reported by @Readon
Assigned to @klenai

numero-744, 2022-10-22:
Introduced by https://github.com/SpinalHDL/SpinalHDL/pull/821/commits/18033becfe533b56b55673c5a34909f3ce940828

Suggested changes:

```diff
+def onWriteByte(address: BigInt, data: Byte): Unit
-def onWriteByte(address: BigInt, data: Byte, id: Int): Unit
+def onWriteByte(address: BigInt, data: Byte, id: Int): Unit = onWriteByte(address, data)
```

Readon, 2022-10-22:
one more thing is about onLast, which has been removed.

numero-744, 2022-10-22:
Meta notes:

* Knowing what these functions do could help during the PR review process, to understand why they can / should (well, in this case, shouldn't) be removed or changed. But they are undocumented :rainbow:
* Breaking API changes without a deprecation period is painful, IMHO more than switching to auto-formatting on a file-by-file basis XD

kleinai, 2022-10-22:
My bad. I had constructed the unburster and ID remover components as part of a larger Axi4 to AxiLite4 change set and I needed a lot more sim agent features. Fix is in PR.

PR #924 (2022-10-22): Axi4 sim agent fixes
Restore original function patterns. Fixes #923 
I needed a few new agent functions to test the unburster (and some subsequently un-PR'd AxiLite4 conversion). I believe these original functions share the same transaction life cycle stage as before.
closed 2022-10-22 by Readon