SpinalHDL/SpinalHDL #929: bug: Test breaks with external soft_reset simulation.

Readon, 2022-10-23:
My test bench can work at 1.7.0 branch well.
However, it breaks when dev branch is using.
I trace it down and write a minimum example there.
Below, code just stalled without error prompt.

``` scala
class softResetTester extends AnyFunSuite {
  var compiled: SimCompiled[Component] = null
  test("compile") {
    compiled = SpinalSimConfig().withFstWave.compile {
      val clockDomain = ClockDomain.external("", withSoftReset = true)
      val area = new ClockingArea(clockDomain) {
        val dut = new Component() {
          val i = in UInt (4 bits)
          val o = out UInt (4 bits)
          val r = RegNext(i)
          r <> o
        }
      }
      area.dut
    }

    compiled.doSim(dut => {
      dut.clockDomain.forkStimulus(10)
      dut.clockDomain.deassertSoftReset()
      dut.clockDomain.waitSampling()
      dut.clockDomain.assertSoftReset()
      dut.clockDomain.waitSampling()
      dut.clockDomain.deassertSoftReset()
      dut.clockDomain.waitSampling()
      simSuccess()
    })
  }
}
```

numero-744, 2022-10-23:
Created new file `tester/src/test/scala/spinal/tester/scalatest/SoftResetTester.scala` which contains:

```scala
package spinal.tester.scalatest

import org.scalatest.funsuite.AnyFunSuite

import spinal.core._
import spinal.sim._

// Should not be needed?
import spinal.core.sim._

class SoftResetTester extends AnyFunSuite {
  var compiled: SimCompiled[Component] = null
  test("compile") {
    compiled = SpinalSimConfig().withFstWave.compile {
      val clockDomain = ClockDomain.external("", withSoftReset = true)
      val area = new ClockingArea(clockDomain) {
        val dut = new Component() {
          val i = in UInt (4 bits)
          val o = out UInt (4 bits)
          val r = RegNext(i)
          r <> o
        }
      }
      area.dut
    }

    compiled.doSim(dut => {
      SimTimeout(10000)
      dut.clockDomain.forkStimulus(10)
      dut.clockDomain.deassertSoftReset()
      dut.clockDomain.waitSampling()
      dut.clockDomain.assertSoftReset()
      dut.clockDomain.waitSampling()
      dut.clockDomain.deassertSoftReset()
      dut.clockDomain.waitSampling()
      simSuccess()
    })
  }
}
```

Then I run:

```bash
sbt clean tester/testOnly spinal.tester.scalatest.SoftResetTester
```

It fails with `SimTimeout` on `dev`, but also on `v1.7.0` :thinking: In both cases if I remove `SimTimeout` it hangs.

<details><summary>Wanted to run this to find the commit which introduced the error</summary>

```bash
git bisect good v1.7.0
git bisect bad dev
git bisect run sbt clean tester/testOnly spinal.tester.scalatest.SoftResetTester
```

</summary>

Dolu1990, 2022-10-24:
Hi,

To me, this test bench should never have worked.

It works with : 

```scala
  compiled.doSim(dut => {
    dut.clockDomain.forkStimulus(10)
    dut.clockDomain.deassertSoftReset()
    dut.clockDomain.waitSampling()
    dut.clockDomain.assertSoftReset()
    dut.clockDomain.waitRisingEdge() //Wait sampling would have wait forever on soft reset deassertion
    dut.clockDomain.deassertSoftReset()
    dut.clockDomain.waitSampling()
    simSuccess()
  })
  ```
  
  Reason that waitSampling is also checking that the softReset is deasserted

Dolu1990, 2022-10-24:
@Readon Your minimal example wasn't stalling on 1.7.0 ?

Readon, 2022-10-24:
> @Readon Your minimal example wasn't stalling on 1.7.0 ?

I did not runthe example there just my testcases

numero-744, 2022-10-24:
This bug report needs more information to reproduce the bug, I update labels.

Readon, 2022-10-24:
I found the problem, thank you all.

closed 2022-10-24 by Readon