SpinalHDL/SpinalHDL #930: BlackBoxing a SystemVerilog module with 2D IO

traxus4, 2022-10-23:
bug report

SystemVerilog have module like this:
```sv
module sh_ddr (
   input[15:0] cl_sh_ddr_awid[2:0],
   input[63:0] cl_sh_ddr_awaddr[2:0],
```
in order to communicate with the said module I wrote a BlackBox definition like this:
```scala
class sh_ddr extends BlackBox {
  val cl_sh_ddr_awid = in Vec (UInt(16 bits), 3)
  val cl_sh_ddr_awaddr = in Vec (UInt(64 bits), 3)
```
When I instantiate the BlackBox SpinalHDL will output SystemVerilog code like this:
```sv
sh_ddr subRam (
    .cl_sh_ddr_awid_0    (crossbar_1_rams_0_aw_payload_id[15:0]  ), //i
    .cl_sh_ddr_awid_1    (crossbar_1_rams_1_aw_payload_id[15:0]  ), //i
    .cl_sh_ddr_awid_2    (crossbar_1_rams_3_aw_payload_id[15:0]  ), //i
    .cl_sh_ddr_awaddr_0  (crossbar_1_rams_0_aw_payload_addr[63:0]), //i
    .cl_sh_ddr_awaddr_1  (crossbar_1_rams_1_aw_payload_addr[63:0]), //i
    .cl_sh_ddr_awaddr_2  (crossbar_1_rams_3_aw_payload_addr[63:0]), //i
```
which does not work because there's no port called `cl_sh_ddr_awid_0` on SystemVerilog side
I think what we need is to handle a special case when BlackBox's IO is Vec of BitVector

numero-744, 2022-10-23:
Does this work?

```scala
object TestThing extends App {
  class Dut extends Component {
    val thing = new SvComponent
    thing.io.a.foreach(_ := 0)
    thing.io.b.foreach(b => assert(b === 0))
  }

  SpinalSystemVerilog(new Dut)
}

class SvComponent extends Component {
  val io = new Bundle {
    val a = in(Vec(Bits(4 bits), 5))
    val b = out(Vec(Bits(4 bits), 5))
  }

  class sv_component extends BlackBox {
    val a = in Bits (20 bits)
    val b = out Bits (20 bits)
  }

  val bb = new sv_component
  bb.a := io.a.asBits
  io.b assignFromBits bb.b
}
```

<details><summary>Generated SystemVerilog</summary>

```verilog
// Generator : SpinalHDL v1.7.3    git head : aeaeece704fe43c766e0d36a93f2ecbb8a9f2003
// Component : Dut
// Git hash  : 181684644c21cb687b259ffca78cbbc6782ca144

`timescale 1ns/1ps

module Dut (
  input               clk,
  input               reset
);

  wire       [3:0]    thing_io_b_0;
  wire       [3:0]    thing_io_b_1;
  wire       [3:0]    thing_io_b_2;
  wire       [3:0]    thing_io_b_3;
  wire       [3:0]    thing_io_b_4;

  SvComponent thing (
    .io_a_0 (4'b0000          ), //i
    .io_a_1 (4'b0000          ), //i
    .io_a_2 (4'b0000          ), //i
    .io_a_3 (4'b0000          ), //i
    .io_a_4 (4'b0000          ), //i
    .io_b_0 (thing_io_b_0[3:0]), //o
    .io_b_1 (thing_io_b_1[3:0]), //o
    .io_b_2 (thing_io_b_2[3:0]), //o
    .io_b_3 (thing_io_b_3[3:0]), //o
    .io_b_4 (thing_io_b_4[3:0])  //o
  );
  always @(posedge clk or posedge reset) begin
    if(reset) begin
    end else begin
      assert((thing_io_b_0 == 4'b0000)) else begin
        $fatal(""); // MyTopLevel.scala:L46
        $finish;
      end
      assert((thing_io_b_1 == 4'b0000)) else begin
        $fatal(""); // MyTopLevel.scala:L46
        $finish;
      end
      assert((thing_io_b_2 == 4'b0000)) else begin
        $fatal(""); // MyTopLevel.scala:L46
        $finish;
      end
      assert((thing_io_b_3 == 4'b0000)) else begin
        $fatal(""); // MyTopLevel.scala:L46
        $finish;
      end
      assert((thing_io_b_4 == 4'b0000)) else begin
        $fatal(""); // MyTopLevel.scala:L46
        $finish;
      end
    end
  end

endmodule

module SvComponent (
  input      [3:0]    io_a_0,
  input      [3:0]    io_a_1,
  input      [3:0]    io_a_2,
  input      [3:0]    io_a_3,
  input      [3:0]    io_a_4,
  output     [3:0]    io_b_0,
  output     [3:0]    io_b_1,
  output     [3:0]    io_b_2,
  output     [3:0]    io_b_3,
  output     [3:0]    io_b_4
);

  wire       [19:0]   bb_a;
  wire       [19:0]   bb_b;

  sv_component bb (
    .a (bb_a[19:0]), //i
    .b (bb_b[19:0])  //o
  );
  assign bb_a = {io_a_4,{io_a_3,{io_a_2,{io_a_1,io_a_0}}}};
  assign io_b_0 = bb_b[3 : 0];
  assign io_b_1 = bb_b[7 : 4];
  assign io_b_2 = bb_b[11 : 8];
  assign io_b_3 = bb_b[15 : 12];
  assign io_b_4 = bb_b[19 : 16];

endmodule
```

</details>

If so, I think it is actually not a bug, more like a missing feature to be able to add conversion (or any) logic into the Blackbox description. Maybe it should be put in a separate unit? As if, in the code above, `SvComponent` was an `Area`, or an "inlined `Component`" :thinking:

numero-744, 2022-10-23:
Eh, tell me, what do you think about this?

```scala
object TestThing extends App {
  class Dut extends Component {
    val thing = new sv_component
    thing.io.a.foreach(_ := 0)
    thing.io.b.foreach(b => assert(b === 0))
  }

  SpinalSystemVerilog(new Dut)
}

class sv_component extends Area {
  val io = new Bundle {
    val a = Vec(Bits(4 bits), 5)
    val b = Vec(Bits(4 bits), 5)
  }

  class sv_component extends BlackBox {
    val a = in Bits (20 bits)
    val b = out Bits (20 bits)
  }

  val bb = new sv_component
  bb.a := io.a.asBits
  io.b assignFromBits bb.b
}
```

It generates this:

```verilog
// Generator : SpinalHDL v1.7.3    git head : aeaeece704fe43c766e0d36a93f2ecbb8a9f2003
// Component : Dut
// Git hash  : 181684644c21cb687b259ffca78cbbc6782ca144

`timescale 1ns/1ps

module Dut (
  input               clk,
  input               reset
);

  wire       [19:0]   thing_bb_a;
  wire       [19:0]   thing_bb_b;
  wire       [3:0]    thing_io_a_0;
  wire       [3:0]    thing_io_a_1;
  wire       [3:0]    thing_io_a_2;
  wire       [3:0]    thing_io_a_3;
  wire       [3:0]    thing_io_a_4;
  wire       [3:0]    thing_io_b_0;
  wire       [3:0]    thing_io_b_1;
  wire       [3:0]    thing_io_b_2;
  wire       [3:0]    thing_io_b_3;
  wire       [3:0]    thing_io_b_4;

  sv_component thing_bb (
    .a (thing_bb_a[19:0]), //i
    .b (thing_bb_b[19:0])  //o
  );
  assign thing_bb_a = {thing_io_a_4,{thing_io_a_3,{thing_io_a_2,{thing_io_a_1,thing_io_a_0}}}};
  assign thing_io_b_0 = thing_bb_b[3 : 0];
  assign thing_io_b_1 = thing_bb_b[7 : 4];
  assign thing_io_b_2 = thing_bb_b[11 : 8];
  assign thing_io_b_3 = thing_bb_b[15 : 12];
  assign thing_io_b_4 = thing_bb_b[19 : 16];
  assign thing_io_a_0 = 4'b0000;
  assign thing_io_a_1 = 4'b0000;
  assign thing_io_a_2 = 4'b0000;
  assign thing_io_a_3 = 4'b0000;
  assign thing_io_a_4 = 4'b0000;
  always @(posedge clk or posedge reset) begin
    if(reset) begin
    end else begin
      assert((thing_io_b_0 == 4'b0000)) else begin
        $fatal(""); // MyTopLevel.scala:L46
        $finish;
      end
      assert((thing_io_b_1 == 4'b0000)) else begin
        $fatal(""); // MyTopLevel.scala:L46
        $finish;
      end
      assert((thing_io_b_2 == 4'b0000)) else begin
        $fatal(""); // MyTopLevel.scala:L46
        $finish;
      end
      assert((thing_io_b_3 == 4'b0000)) else begin
        $fatal(""); // MyTopLevel.scala:L46
        $finish;
      end
      assert((thing_io_b_4 == 4'b0000)) else begin
        $fatal(""); // MyTopLevel.scala:L46
        $finish;
      end
    end
  end

endmodule
```

numero-744, 2022-10-23:
So your example could start like:

```scala
class sh_ddr extends Area {
  val io = new Bundle {
    val cl_sh_ddr_awid = Vec(UInt(16 bits), 3)
    val cl_sh_ddr_awaddr = Vec(UInt(64 bits), 3)
  }

  class sh_ddr extends BlackBox {
    val cl_sh_ddr_awid = in Bits (3 * 16 bits)
    val cl_sh_ddr_awaddr = in Bits (3 * 64 bits)
  }

  val bb = new sh_ddr
  bb.setPartialName("")

  bb.cl_sh_ddr_awid := io.cl_sh_ddr_awid.asBits
  bb.cl_sh_ddr_awaddr := io.cl_sh_ddr_awaddr.asBits
}
```

numero-744, 2022-11-07:
No activity since suggested solution 15 days ago, I am closing this issue. @traxus4 feel free to re-open it if the suggested solution does not fix your needs.

I link with https://github.com/SpinalHDL/SpinalDoc-RTD/issues/131 because it might be interesting (open to discussion)

closed 2022-11-07 by numero-744