SpinalHDL/SpinalHDL #934: IDEA compile failed

balanx, 2022-10-24:
The following is error information.

[image]

numero-744, 2022-10-24:
@Dolu1990

Readon, 2022-10-24:
Which version of Scala are you using?
I can find the Info in "core/target/scala-2.11/src_managed/main/Info.scala", which means it is a class from outside SpinalHDL.

numero-744, 2022-10-24:
@balanx It popped on my screen but after a clean it has gone. Have you tried to clean the workspace?

balanx, 2022-10-24:
@Readon  My scala version is 2.12.16; IDEA is 2022.2.2
@numero-744 How to clean ? I deleted '.idea/*'  'target/*' and rebuilded the project, it has NOT gone.

numero-744, 2022-10-25:
Okay the error is back on my side with VSCodium. I think it is linked to having both Metals building and `sbt` cleaning, leading to inconsistencies?

Metals "clean compile workspace" does not seem to work and `sbt clean` too, I don't know how Metals work actually XD.

What I did is actually **something I do not recommend** :warning: because it deletes all un-commited things, even ignored things:

```bash
git clean -fdx
```

It is like removing the whole folder and re-cloning (but you keep the local history).

Metals reports using Scala 2.11.12, I have not checked when I had the error.

I have set `build.sbt` to use Scala 2.12.13 now, no errors yet.

@Readon Info is generated by [`build.sbt`](https://github.com/SpinalHDL/SpinalHDL/blob/dev/build.sbt#L140).

@balanx Can you try to close IDE, remove `target/` and `core/target/`, re-open IDE and tell me if it works, please?

balanx, 2022-10-25:
@numero-744 There is No error after I did what you said "try to close IDE, remove .idea/*, target/ and core/target/, re-open IDE".
Thanks so much.

closed 2022-10-25 by numero-744

Dolu1990, 2022-10-25:
Ahhh i often have this kind of stuff in intellij XD
real pain in the ass.
What i generaly do is to do a sbt clean compile, and then use the IDE