SpinalHDL/SpinalHDL #947: Some questions about `SFix` bitwidth adaptation

dzx-dzx, 2022-11-05:
The `accumulator`, `weight` and `activation` are all defined as `SFix(peak = 8 exp, width = 16 bits)`.
1. Why does the following:
```scala
accumulator := accumulator + (io.weight * io.activation).truncated
```
throw the following error:
```
[error] toplevel/accumulator : SFix[peak=2^8 resolution=2^-7] can't be assigned by toplevel/??? : SFix[peak=2^17 resolution=2^-14] because of truncation. You can do x := y.truncated if that's fine.
```
...but this:
```scala
accumulator := (accumulator + io.weight * io.activation).truncated
```
is fine?
2. Why does
```scala
accumulator := io.clear ? operandType().getZero | (accumulator + io.weight * io.activation).truncated
```
throws
```
[error] WIDTH MISMATCH (16 bits <- 32 bits) on (toplevel/??? :  SInt[16 bits]) := (Bool ? Bits | Bits)[32 bits]
```
?

andreasWallner, 2022-11-05:
The idea behind `truncated` is that the result is flagged for a possible truncation, but that truncation can only happen when assigned to a variable (which is narrower). When you add `accumulator` to the truncated product this flag is not carried along, therefore the result of the addition can't be assigned. With your different parenthesis the final result is flagged to be truncated, so that's fine. - Maybe the API can be improved here by adding a `truncate` method that would already to the truncation and return the smaller value (similar to `resize` and `resized` for `Bits`)

With your second question I think that is actually a bug, since here: https://github.com/SpinalHDL/SpinalHDL/blob/155ff09204c4a5c1dfbb33c793e08991afa1a1ce/core/src/main/scala/spinal/core/internals/Node.scala#L127 only `BitVector` is resized, and `SFix` does not derive from `BitVector`. I can have a look at fixing this, or do you want to give it a go?

dzx-dzx, 2022-11-05:
> With your second question I think that is actually a bug, since here:

Oh yes, thanks!

andreasWallner, 2022-11-09:
I had a bit more detailed look at this, and there are a few other pitfalls there - not the least that we actually never specified what `truncate` should really do... I'll continue to have a look, for now I'll open another issue to discuss that point.

In the meantime I'd recommend to assign to an intermediate signal and use that in the multiplexer.

dzx-dzx, 2022-11-09:
> In the meantime I'd recommend to assign to an intermediate signal and use that in the multiplexer.

I see, thanks.

commit 8188f9df (2022-11-28): Fix XFix multiplexer usage

Fixes #947
closed 2022-11-28 by Dolu1990