SpinalHDL/SpinalHDL #948: Test seed not being randomized anymore is confusing

andreasWallner, 2022-11-05:
In this change: https://github.com/SpinalHDL/SpinalHDL/commit/1e94dd2efd95e36036ec9f1789e9d6d11442e57f, because of #657 you changed the randomization behavior in the simulation.
To me this is quite a misleading behavior, I (and some colleagues of mine) assumed that tests are randomized if I don't pass a fixed seed to them. We expected a call to `randomize` would randomize by default, 
It also makes randomized tests in e.g. intellij much more inconvenient, since I can't use the "Play" button beside a test anymore to run it with random seed, except if I started intellij from a command line with the variable set (or it being set globally) - and not from the start menu. It's similar with VSCode & Metals. If one needs a fixed seed it's much easier by just changing the `seed` parameter.

Please reconsider the change, to me it feels like a regression.

If not, we should document the behavior in RTD, the sim environment variables are currently not mentioned there. We might also want to change the SpinalHDL github actions to set this environment variable?

numero-744, 2022-11-05:
Oooh I did not notice the change. What exactly is the changed behavior?

Does it imply initial values of signals? Does it imply values chosen with `Random` and `randomize()` API?

Side note: pushing on `dev` or `master` does not seem a nice thing, doing PRs is good to discuss this kind of things ^^'

kleinai, 2022-11-06:
Yes this behavior should be rolled back

Dolu1990, 2022-11-06:
So, let's change 

```scala
def doSimApi(name: String = "test", seed: Int = sys.env.getOrElse("SPINAL_SIM_SEED_RANDOM", "0").toInt*Random.nextInt(2000000000),
```

Into : 
```scala
  def newSeed(): Unit ={
    sys.env.get("SPINAL_SIM_SEED") match {
      case Some(v) => v.toInt
      case None => sys.env.getOrElse("SPINAL_SIM_SEED_RANDOM", "1") match {
        case "1" => Random.nextInt(2000000000)
        case "0" => 0
      }
    }
  }
def doSimApi(name: String = "test", seed: Int = newSeed(), ...)
```

?

That way we provide a way to fix the seed easily without code modification. Also if a regression fail, you can fix the seed via environnement variable

andreasWallner, 2022-11-06:
I'm not sure I understand, that would mean:
- setting the seed parameter overrides everything else
- setting SPINAL_SIM_SEED in the terminal allows to fix the seed to a certain value
- setting SPINAL_SIM_SEED_RANDOM to 0 uses seed 0
- setting SPINAL_SIM_SEED_RANDOM to 1 uses a random seed
- setting SPINAL_SIM_SEED_RANDOM to anything else gives a match error
- setting nothing also gives a random seed

I'm not sure I follow why you'd include both SPINAL_SIM_SEED_RANDOM and SPINAL_SIM_SEED you can achieve the same result with only one of them? (okay SSS can do slightly more than SSSR)
Do you want to keep SPINAL_SIM_SEED_RANDOM for backwards compatibility - it still kind of breaks previous behavior with changing the default?

PS: I totally get the need why it's nice to fix the seed w/o code changes, that was actually on my list of PRs to make ;-)

Dolu1990, 2022-11-07:
@andreasWallner Yes you got it right ^^ 

> I'm not sure I follow why you'd include both SPINAL_SIM_SEED_RANDOM and SPINAL_SIM_SEED you can achieve the same result with only one of them?
> Do you want to keep SPINAL_SIM_SEED_RANDOM for backwards compatibility

Yes exactly ^^

> it still kind of breaks previous behavior with changing the default?

Which default / what default ? (in which context ?) Sorry i'm not sure exactly XD

> PS: I totally get the need why it's nice to fix the seed w/o code changes, that was actually on my list of PRs to make ;-)

Ahhh nice :D So i haven't pushed anything, feel free to do so if you thinkg that the proposal is good ^^

andreasWallner, 2022-11-08:
What I meant:

If we currently have a user who does not want randomization, they'll most likely not define the environment variable, this change would break their setup. 
If we have a user that wants randomization they will have the env variable, which could be set to any non-zero value - so there is at least a good change to break their setup (I used 42 just because...).

So either way we break many setups, therefore I would argue we could break a few more and go to the simple solution with only a single env var SSS. Less to document/explain/understand.

I'd be happy to open a PR, but that will have to come after #947 ;-)

Dolu1990, 2022-11-09:
So, i think the reason why the seed was changed to be fixed, is that the "most common case" is you write down a testbench, it will obviously fail a few times, and you will anyway to have a fixed seed to re-run it in the same conditions to be able to compare.

Having randomized seed may be only usefull in the context you want to run a tests multiple times with multiple seeds.

That's kind of a hard to balance situation.

But i would say, in some ways, when i run a program or something, i want a deterministic / reproducable behaviour, unless explicitly specifed. I think that is kind of the main argument against default randomisation of the seed.

Would it make sense XD ?

andreasWallner, 2022-11-09:
For me the flow is mostly:
- write a new testcase
- run simulations
- set a specific seed that triggered some issue
- repeat until done ;-)
- remove fixed seed for CI and normal tests to have a chance catching stuff that happens infrequently

I guess if I want a deterministic test then I should not use `randomize` ;-)

But my point was just that I think we could do with a single env var instead of both, also you example would enable randomness by default again?

Dolu1990, 2022-11-10:
> I think we could do with a single env var instead of both

Yes ^^

> your example would enable randomness by default again?

Yes

I don't know what is the default behaviour of GHDL / IVerilog / Verilator / and others about it. If the default seed is randomized or not XD
Any experience on it ?

andreasWallner, 2022-11-20:
Just to not loose the info, as posted on the Gitter channel

- VHDL doesn't have an RNG with a hidden seed, so it doesn't really compare (you always have to provide a seed, regardless what you want)
- Verilog: the LRM seems to leave that open
  - Verilator: has two randomizations (;-))
    - the initial state of a wire is "true" random by default(its just never initialized if not requested)
    - verilog $random is seeded using the system RNG
- cocotb uses a random seed and has a RANDOM_SEED command line option to fix the seed to a specific value (like you proposed as SPINAL_SIM_SEED)
- nmigen doesn't seem to have a generator, just using the python rng - so it depends on the test framework used

So we'd be in good company with a default-random seed - I'll go ahead an open a PR for that.

closed 2022-11-21 by Dolu1990