SpinalHDL/SpinalHDL #949: Is there a way to define a `clear` signal that behaves much like the `reset` signal, but can be driven through custom logic?

dzx-dzx, 2022-11-07:
I worry that some residue data in, for example, a group of shift registers created from `History` will produce unexpected behavior. But I'm at a loss to find a way to reset a module.

Readon, 2022-11-07:
You can have an internal "ResetArea" in your design. I think it's described in the documentation.
Or create a ClockingArea with softreset.

dzx-dzx, 2022-11-08:
Oh thanks! I'd give it a try.

dzx-dzx, 2022-11-08:
@Readon I took a look at the compiled result… Is it sensible to decorate a custom (synchronous?) signal with `negedge`?

Readon, 2022-11-09:
> 

If you want a sync version, then you can use a softreset to do that.

I did not have your test code, so I guess you are using a reset to do that.

Might be worth to try below
``` scala
val area = ClockDomain.current.copy(softReset = io.clear) on {
   ....
}
```
If I remember right.

dzx-dzx, 2022-11-09:
Ah thanks! This does work.

closed 2022-11-09 by dzx-dzx