SpinalHDL/SpinalHDL #951: RTL assertion works before reset

RicardoNid, 2022-11-08:
even though I assertReset at the very beginning of simulation, when there're RTL assertions in the DUT, these assertions may be evaluated before reset is asserted(like in the waveform attached, the assertion require that badValid1 shouldn't never be true), for DUTs without assertions, everything works fine, any workaround?
I tried to use reset/clk as part of the assertion, but reset/clk of the default clockDomain seem to be not accessible in the Component
[image]

andreasWallner, 2022-11-08:
You can access the ClockDomain via `ClockDomain.current`. I think there is currently no 'clocked' assertion, but I might have overlooked stuff from the formal verification parts.

numero-744, 2022-11-08:
Hmm I thought all Spinal assertions were performed upon sampling?

andreasWallner, 2022-11-08:
@numero-744 you are right, I had forgotten about that.

Readon, 2022-11-09:
There is also no clock there.

BTW, it might not be a formal verification case, because there is no simulation in that. 
"assertion" for simulation and for formal verification are always has the same name.

wswslzp, 2022-11-10:
Does the Spinal Formal generate the SVA statements? I don't see any example that generates SVA in the documents. https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Formal%20verification/index.html

Readon, 2022-11-10:
> 

Sequences and properties are not supported now, because free version of SymbiYosys does not do so.
Adding SpinalHDL native support by FSM is possible, but it would be a duplication of wheel building.

wswslzp, 2022-11-10:
I don't see the implication operations like `|->` or `|=>` in SVA. But I see `when` appears in the doc. So you construct similar assertions using `always` in Verilog? Pls show some generated Verilog/SV codes in your documents.

Readon, 2022-11-10:
Yes, you are right there is no SVA support at all, all supporting is based on Verilog compliant version.

closed 2022-11-21 by numero-744