SpinalHDL/SpinalHDL #954: Some more questions about `SFix` - and its cousins, `SInt` and `SQ`

dzx-dzx, 2022-11-10:
I'm attempting to employ the clamping technique in my fix-point project. I'm afraid... I'm a bit confused by the result I found.
From [this page](https://spinalhdl.github.io/SpinalDoc-RTD/dev/SpinalHDL/Data%20types/Int.html#fixpoint-operations) and PR #233 I understand that `SInt` suppports fix-point operations. And according to #422 and #285 this is actually preferred way. But also in #422 the use of `SQ` is briefly mentioned. 

What are the relations among them? Why is `SFix` less welcome?

wswslzp, 2022-11-10:
I suggest that you should do some experiments before using the SInt/SFix/AFix rounding. Make sure you know the actual behavior. 

SFix/UFix just has some basic arithmetic operations without rounding methods. AFix is supposed to replace SFix/UFix, which unifies the SFix and UFix and adds all kinds of rounding operations. Also, these fixed-point types can track the value range and report overflow/underflow errors. It's a little bit HLS style. 

SInt/UInt also has fixed point rounding operations but no range check as SFix/UFix/AFix does. SQ/UQ is Q format for SInt/UInt respectively, which carry the fixed-point position. This is a traditional RTL style, where you already have a fixed-point C algorithm model.

dzx-dzx, 2022-11-10:
So `AFix` will hopefully be the ultimate choice. And... `SFix` *is* stored as Q format, while SQ builds upon `SInt` and *implements* Q format?

kleinai, 2022-11-11:
I wrote AFix mostly as a replacement of SFix and UFix. The existing API didn't lend itself to operations between the types. AFix will automatically scale as necessary, allows for mixed fixed and unfixed math, and has slightly more rounding modes.

dzx-dzx, 2022-11-11:
> I wrote AFix mostly as a replacement of SFix and UFix.

Is there a scenario where neither matches your need?

kleinai, 2022-11-11:
Not that I've found yet. It should be a comprehensive replacement but I don't think it has adoption yet. From where I've used it it's done what I need.

dzx-dzx, 2022-11-11:
OK, thanks!

closed 2022-11-11 by dzx-dzx