SpinalHDL/SpinalHDL #955: Different result about getWidth

kdurant, 2022-11-11:
```scala
  val axis_s =  Stream(Fragment(Bits(32 bits)))
  println(axis_s.payload.getWidth) 
  println(axis_s.payload.asBits.getWidth)
```

`println(axis_s.payload.getWidth) ` output is 32
`println(axis_s.payload.asBits.getWidth) ` output is 32

numero-744, 2022-11-11:
What happens is that:

* `axi_s.payload` is a `Fragment[Bits]`, which is a `payload = Bits(32 bits)` and a `last = Bool`
* `.getWidth` is defined for `BitVector` (so it is a member of `Bits`) but not for `Fragment`.
* there is an implicit conversion from `Fragment[T]` to `T`, so here for `Fragment[Bits]` to `Bits`, and the conversion is just returning the payload
* so Scala performs the implicit conversion for `.getWidth`, so the first line is equivalent to `println(axis_s.payload.payload.getWidth)`, hence the result which is 32.
* the second concatenates payload & last, which results in 33 bits.

The second line is equivalent to `println(axis_s.payload.getBitsWidth)` and `println(widthOf(axis_s.payload))`, which work for any SpinalHDL type (representing a hardware signal).

An issue has been open to document this behavior: https://github.com/SpinalHDL/SpinalDoc-RTD/issues/140

wswslzp, 2022-11-12:
The implicit conversion may be confusing in cases like this. Is this necessary?

kleinai, 2022-11-12:
I think this is due to the way bundles get initialized. It might be hard to remove due to existing code. At least I see this as a feature but it can be confusing.

closed 2022-11-20 by numero-744