SpinalHDL/SpinalHDL #956: Suggestion: Bus monitor direction

kleinai, 2022-11-11:
Does Spinal have a monitor bus type? It could be useful for in-system bus debug components.

Maybe something like this is sufficient?
```scala
object monitor extends MS {
  def apply[T <: IMasterSlave](i: T) = {
    i.asInstanceOf[Data].asInput()
    i.isMasterInterface = false
    i
  }
}
```

numero-744, 2022-11-11:
There is not (yet). Is it really useful in hardware description? Not just a simulation thing?

kleinai, 2022-11-11:
I've used Xilinx bus monitor IP in hardware

numero-744, 2022-11-11:
Oh, was it on a blackbox?

I can PR this feature quickly if others agree we should implement it (and document it later™)

Readon, 2022-11-11:
Is it an online monitor logic that inject into the design?

kleinai, 2022-11-11:
Like a stream monitor to capture data for debug.

I've made a PR #957

numero-744, 2022-11-11:
While reviewing the PR, I have found this line which triggered thinking:

[`def asMonitor(): Unit = this.asInstanceOf[Data].asInput()`](https://github.com/SpinalHDL/SpinalHDL/pull/957/files#diff-0d895d626825770c57d288b933d6deaf6eedbbf2f67105a2ca1bbed11b485457R74)

What more can we expect from this function? I mean, this set of `asXxx` functions is meant to be overridden with directions which are protocol-specific. Here I am not sure it is useful, maybe `monitor` should just set inputs.

Also [line 47](https://github.com/SpinalHDL/SpinalHDL/pull/957/files#diff-0d895d626825770c57d288b933d6deaf6eedbbf2f67105a2ca1bbed11b485457R47) it is defining itself as a slave, which is not really true.

So here is my opinion on this topic: maybe, instead of adding a word in the "bus" vocabulary, `in` should be used? It is still available, and seems to represent what you want. No need to PR and update documentation, just know that the solution already exists.

```scala
val io = new Bundle {
  val monitored = in port MyBus(arguments)
}
```

kleinai, 2022-11-11:
I rescind my PR. `in(...)` does work as intend. I thought for sure this wouldn't work in the past.

closed 2022-11-11 by kleinai

numero-744, 2022-11-11:
Anyway thanks for the PR @kleinai, it was useful to have concrete things to think about!