SpinalHDL/SpinalHDL #961: How to concatenate the same data together many times?

AlbertGeorge42, 2022-11-16:
I want to concatenate a same data many times like verilog "assign a = {5{b}};".
I try to fill the data into the List as parameters and use the Cat function to do the job.
  val b = Bool()
  val bb = List.fill(5)(b)
  val a = Cat(bb:_*)
Is there a corresponding function in SpinalHDL?

andreasWallner, 2022-11-16:
There is the (I think undocumented) operator `#*` (https://github.com/SpinalHDL/SpinalHDL/blob/ad698c3171c5f0287a26ab5d4d15488ec525ae9a/core/src/main/scala/spinal/core/Data.scala#L735) on `Data` which is implemented with the code that you have shown. That should do what you need.

Alternatively you could use `B(5 bits, default -> x)` if the value is know at compile time.

AlbertGeorge42, 2022-11-16:
Thanks for the answers, I also didn't find the #* operation in the document.

closed 2022-11-16 by AlbertGeorge42

numero-744, 2022-11-16:
Yea, it will be added https://github.com/SpinalHDL/SpinalDoc-RTD/issues/142