SpinalHDL/SpinalHDL #963: StreamFragmentWidthAdapter(x, y, earlyLast = true) widthOf(y) = 2 *widthOf(x) triggers assertion in whenIndexed()

likewise, 2022-11-16:
```scala
// minimal case to reproduce bug for case StreamFragmentWidthAdapter(x, y = x * 2, earlyLast = true)
case class AxisUpsizer(dataWidthIn : Int, dataWidthOut: Int) extends Component {
  require(dataWidthOut == (2 * dataWidthIn), "dataWidthOut must be a factor 2 of dataWidthIn")
  require(dataWidthOut % dataWidthIn == 0, "dataWidthOut must be an integer multiple of dataWidthIn")

  val io = new Bundle {
    val sink = slave Stream(Fragment(Bits(dataWidthIn bits)))
    val source = master Stream(Fragment(Bits(dataWidthOut bits)))
  }
  StreamFragmentWidthAdapter(io.sink, io.source, earlyLast = true)
}
```
(or any factor 2 with earlyLast = true) errors out as follows:
```scala
[error] Exception in thread "main" java.lang.AssertionError: assertion failed
[error]         at scala.Predef$.assert(Predef.scala:156)
[error]         at spinal.core.package$.assert(core.scala:463)
[error]         at spinal.lib.whenIndexed$.apply(Utils.scala:1320)
[error]         at spinal.lib.StreamFragmentWidthAdapter$$anon$56$$anon$41$$anonfun$41.apply$mcV$sp(Stream.scala:1601)
[error]         at spinal.core.when$.apply(when.scala:90)
[error]         at spinal.lib.StreamFragmentWidthAdapter$$anon$56$$anon$41.<init>(Stream.scala:1600)
[error]         at spinal.lib.StreamFragmentWidthAdapter$$anon$56.<init>(Stream.scala:1562)
[error]         at spinal.lib.StreamFragmentWidthAdapter$.apply(Stream.scala:1541)
[error]         at corundum.AxisUpsizer.<init>(AxisUpsizer.scala:84)
[error]         at corundum.AxisUpsizerSim$$anonfun$main$1.apply(AxisUpsizerSim.scala:18)
[error]         at corundum.AxisUpsizerSim$$anonfun$main$1.apply(AxisUpsizerSim.scala:18)
[error]         at spinal.core.internals.PhaseCreateComponent$$anonfun$impl$66.apply(Phase.scala:2404)
[error]         at spinal.core.fiber.Engine$$anonfun$create$1.apply$mcV$sp(AsyncCtrl.scala:147)
[error]         at spinal.core.fiber.AsyncThread$$anonfun$1.apply$mcV$sp(AsyncThread.scala:59)
[error]         at spinal.core.fiber.EngineContext$$anonfun$newJvmThread$1.apply$mcV$sp(AsyncCtrl.scala:39)
```

Root cause analysis (also see Gitter) is that ```buffer``` stores one input word, whereas the counter counts in [0,1] range, thus the number of bits required for indexes differs in the whenIndexed(), triggering the assertion.

Fix could be to use relaxedWidth = (factor <= 2) in 

```scala
          when(input.fire) {
            whenIndexed(buffer.subdivideIn(inputWidth bits), counter, relaxedWidth = (factor <= 2)) {
              _ := input.fragment.asBits
            }
          }
          whenIndexed(data.subdivideIn(inputWidth bits).dropRight(1), counter, relaxedWidth = (factor <= 2)) {
            _ := input.fragment.asBits
          }
```

likewise, 2022-11-16:
tested against dev branch, commit ad698c3171c5f0287a26ab5d4d15488ec525ae9a

kleinai, 2022-11-17:
Semi-related, I see you have Axis in your function. You may be interested in: https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/bus/amba4/axis/Axi4StreamSimpleWidthAdapter.scala

closed 2022-11-21 by Dolu1990

likewise, 2022-11-24:
> Semi-related, I see you have Axis in your function. You may be interested in: https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/bus/amba4/axis/Axi4StreamSimpleWidthAdapter.scala

I started creating a bunch of Axis components when you didn't start yet in lib. Would be happy to merge mine into your lib work.

I also want to create a TAP interface that drives a AXIS interface. I have the Java code but lack the Scala experience to port it over quickly.

Dolu1990, 2022-11-28:
>  I have the Java code but lack the Scala experience to port it over quickly.

I'm not sure to understand, the hardware is implemented in java  ?

likewise, 2022-11-28:
Unrelated to this (now correctly closed) issue Inwas explaining I could merge my AXIS components as well as try to port a TUNTAP server (to AXIS) which I have found written in Java.

My remark was only about the TUNTAP server.

PR #964 (2022-11-21): fix: StreamFragmentWidthAdapter(x, y, true) widthOf(y) = 2 * widthOf(x)
Root cause is that counter is in range [0,1], number of elements in buffer Vector is 1, thus the number of bits differs in whenIndexed(), only for the case of widthOf(y) = 2 * widthOf(x). Triggering assert in whenIndexed().

Closes #963

# Context, Motivation & Description

# Impact on code generation

# Checklist

- [x] Unit tests were added
- [x] API changes are or will be documented:
  - using Scaladoc comments: `/** */`?
  - on [RTD](https://github.com/SpinalHDL/SpinalDoc-RTD)?
  - thanks to a [new tracking issue on RT [... cut]