SpinalHDL/SpinalHDL #978: `BigInt().toBytes(...)` is inconvenient to use/broken(?) for positive numbers with "sign bit" set

andreasWallner, 2022-11-27:
When using the `toBytes` method coming from `SimBigIntPimper` the following behavior surprised me:

``` scala
// fails with assertion due to internal length check
// assertion failed: Original BigInt has more bytes then bits specified.
BigInt(0xFF).toBytes(8, endian=BIG) 
// returns [0x00, 0xff]
BigInt(0xFFFFFFFFL).toBytes(endian=BIG)
```
Due to the sign bit being output when calling `BigInt.toByteArray`.

The same issue occurs in the usage locations of `toBytes` in the library, e.g.:
```scala
class Playground1 extends AnyFunSuite {
  case class test01() extends Component {
    val io = new Bundle {
      val u8 = in UInt (8 bit)
    }
  }
  test(s"blub") {
    SimConfig.compile(test01()).doSim("blub") { dut =>
      dut.io.u8 #= 0xff
      sleep(10)
      println(dut.io.u8.toBytes mkString(" "))
    }
  }
}
```
Aborts the simulation because the same assertion as mentioned above fails.
A similar case can be constructed for the use of it in the `SimBitVectorPimper`.

From the usages I'd say that it looks like we wanted the method to get the bit as they are. In that sense I'd "fix" this by ignoring an initial 0 byte if it is there.
That might lead to some surprises when doing something like (just to illustrate) `S(x, 9 bit).toBigInt().toBytes(8)` which would work as long as the number is positive, get an error if negative. But that is no so different from `U(x, 32 bit).toBigInt().toBytes(8)` which will work as long as the number is small enough.
It would also mean that as long as numbers are positive we'd remove a "leading sign bit 0", but this is actually the behavior that we need for e.g. the simulation example above...

Any thoughts? (I'll open a PR with test that show the implications)

While on the topic: Another thing that comes to mind looking at the code is that while we pass a length in bits to the method, it will internally round up to full bytes which seems a bit inconsistent. It might make sense to check for the actually needed length in bits?

Readon, 2022-11-28:
I have tested all cases you give.

Firstly, the cases fail are all with default bits with -1, which is only return the default Bytes that BigInt provided.
Secondly, it's a toBytes method which returns bytes but bits actually, the reason to leave the paramenter with bits unit is that getBitsWidth is provided but bytes. But it could be improved.

The default toByteArray really do some extension when the value is negtive. 
I don't know if it is caused by Int to BigInt conversion or toByteArray method.

andreasWallner, 2022-11-28:
@Readon The issue does not only occur with 0xff, also with all other values that have the MSBit of the MSByte set.
What I see as the problem here is that the value is not -1, but +255. As shown in my second example this occurs when converting an unsigned type to BigInt (which actually does the right thing w.r.t. the sign of the number, as it's unsigned per definition), and then on to bytes which will fail. No negative numbers are involved in the process.

The extension that happens is that toByteArray is implemented to generate the byte representation for a number that might be signed, so it has to do the extension for positive numbers (with 0, otherwise a 255 would be indistinguishable from -1). But that behavior is quite surprising in the SpinalHDL context when coming from e.g. a `UInt` type (an information that we sadly loose).

Notice also that in my second example the user never converts to a BigInt, that happens behind the scenes and seems just broken to me the way it is.

Ad bytes vs. bits: Yes it's `toBytes` which makes sense since it returns an array of bytes, but I think we are on the same page that it's weird to round up internally after having a `bits` parameter - but that is the smaller issue anyways.

kleinai, 2022-11-28:
I had done some testing of BigInt last time this was an issue:
https://github.com/SpinalHDL/SpinalHDL/pull/898#issuecomment-1279525281

PR #979 (2022-12-05): fix: SimBigIntPimper toBytes
Fix zero sign bit being reserved, leading to errors when converting values with MSB of high byte being set (adding an unexpected byte)

Fixes #978

I kind of dislike the helper function in the Unittests, but w/o it they become quite unreadable IMHO...

# Checklist

- [x] Unit tests were added
- Simulation helper are currently undocumented
commit 167d6c33 (2022-12-05): Fix SimBigIntPimper toBytes

Fix zero sign bit being reserved, leading to errors when converting
values with MSB of high byte being set (adding an unexpected byte)

Fixes #978
closed 2022-12-05 by Readon