SpinalHDL/SpinalHDL #986: Sim: support "force" in component internals

numero-744, 2022-12-05:
It can be useful to help the users simulate fault injections:

Implementation examples by Dolu1990 (via the Matrix channel)

```scala
  def simBypass(that : UInt) = new Area{
    val bypassEnable = RegInit(False) simPublic()
    val bypassValue = Reg(that) simPublic()
    when(bypassEnable){
      that := bypassValue
    }

    bypassEnable := bypassEnable
    bypassValue := bypassValue

    def force(value : Int) = {
      bypassEnable #= true
      bypassValue #= value
    }
  }

  SimConfig.withFstWave.compile(new Component {
    val counter = Reg(UInt(8 bits)) init(0) simPublic()
    counter := counter + 1
    val result = out(CombInit(counter))

    val bypass = simBypass(counter)
  }).doSim{ dut =>
    dut.clockDomain.forkStimulus(10)
    dut.clockDomain.waitSampling(5)

    println(dut.counter.toInt)
    sleep(1)
    println(dut.counter.toInt)
    dut.bypass.force(100) //THIS THIS THIS
    println("XXX")
    sleep(1)
    println(dut.counter.toInt)
    dut.clockDomain.waitSampling(1)
    println(dut.counter.toInt)
    dut.clockDomain.waitSampling(1)
    println(dut.counter.toInt)
  ...
```

```scala
  import spinal.core.sim._

  case class BypassTag(enable : Bool, value : UInt, bypass : UInt) extends SpinalTag
  def simBypass(that : UInt) = new Area{
    val bypassEnable = in(Bool())
    val bypassValue = in(cloneOf(that))
    val bypass = cloneOf(that)

    that.addTag(new BypassTag(bypassEnable, bypassValue, bypass))

    def force(value : Int) = {
      bypassEnable #= true
      bypassValue #= value
    }
    def release(): Unit ={
      bypassEnable #= false
    }
  }

  SimConfig.withConfig(
    SpinalConfig().addTransformationPhase(new Phase{
      override def impl(pc: PhaseContext) = {
        pc.walkStatements{ s =>
          // Swap usages of the signal
          s.walkRemapDrivingExpressions{
            case bt : UInt => bt.getTag(classOf[BypassTag]) match {
              case Some(tag) => tag.bypass.pull()
              case None => bt
            }
            case e => e
          }

          //Assign the swaped signal with the swap value (usefull for registers)
          s match {
            case bt : UInt  =>
              bt.getTag(classOf[BypassTag]) match {
                case Some(tag) => bt.component.rework{
                  when(tag.enable.pull()){
                    bt := tag.value.pull()
                  }
                }
                case None =>
              }
            case _ =>
          }
        }

        // Implement the bypass muxes
        pc.walkStatements {
          case bt : UInt  =>
            bt.getTag(classOf[BypassTag]) match {
              case Some(tag) => bt.component.rework{
                tag.bypass := tag.enable.pull() ? tag.value.pull() | bt
              }
              case None =>
            }
          case _ =>
        }
      }
      override def hasNetlistImpact = true
    })
  ).withFstWave.compile(new Component {
    val counter = Reg(UInt(8 bits)) init(0) simPublic()
    counter := counter + 1
    val result = out(CombInit(counter))
    val bypass = simBypass(counter)
  }).doSim{ dut =>
    dut.bypass.release()
    dut.clockDomain.forkStimulus(10)
    dut.clockDomain.waitSampling(5)

    println(dut.result.toInt)
    sleep(1)
    println(dut.result.toInt)
    dut.bypass.force(100) //THIS THIS THIS
    println("XXX")
    sleep(1)
    println(dut.result.toInt)
    dut.clockDomain.waitSampling(1)
    println(dut.result.toInt)
    dut.clockDomain.waitSampling(1)
    println(dut.result.toInt)
    dut.clockDomain.waitSampling(1)
    println(dut.result.toInt)

}
```

xiaochuang-lxc, 2023-08-05:
All EDA Support Force/Relase semantics. Is it better to support it by VPI?
@Dolu1990

Dolu1990, 2023-08-05:
I don't know if Verilator has a force sementic, i would say it hasn't.
I kinda would prefer having that feature being suported the emulated way as shown in the first post, to keep things very portable.

xiaochuang-lxc, 2023-08-05:
It support .
I prefer using Force/Release just like in verilog/systemVerilog/cocotb,and the above way can't use if i want to force signal in an blackbox

Dolu1990, 2023-08-05:
How verilator support it ?
via (* verilator public *) ?

xiaochuang-lxc, 2023-08-05:
/*verilator&32;forceable*/
Specifies that the signal (net or variable) should be made forceable from C++ code by generating public <signame>__VforceEn and <signame>__VforceVal signals. The force control signals are created as
public_flat signals.
To force a marked signal from C++, set the corresponding __VforceVal variable to the desired value, and the
__VforceEn signal to the bit-mask indicating which bits of the signal to force. To force all bits of the target signal, set __VforceEn to all ones. To release the signal (or part thereof), set appropriate bits of the __VforceEn
signal to zero.
Same as forceable in configuration files.

xiaochuang-lxc, 2023-08-05:
The Verilator Document:https://veripool.org/ftp/verilator_doc.pdf

Dolu1990, 2023-08-05:
Hooo lol, nice ^^

xiaochuang-lxc, 2023-08-26:
@Dolu1990 Will the Force/Release support in SpinalHDL? I'm not familiar with Simulator VPI

Dolu1990, 2023-08-27:
i'm kind of a bit scared adding support directly on the simulation backends XD
Would kinda prefer adding it the "soft" way, which would also have the advantage to make it visible in simulation waves

I very very busy for the next 2 months, but will try to find some time to try

xiaochuang-lxc, 2023-08-27:
Recently, I used a lot of Force and Release in cocotb during work, but the simulation speed of cocotb is far less than SpinalHDL. This is not urgent, but it may be a good idea to provide an API that can be integrated by various emulators. At present, the simulation speed of cocotb is far from meeting the needs of system simulation, and SpinalHDL may be a better tool in the future. :)

Readon, 2023-08-27:
> I don't know if Verilator has a force sementic, i would say it hasn't. I kinda would prefer having that feature being suported the emulated way as shown in the first post, to keep things very portable.

it seems that verilator supported force/release on 4.218.
And GHDL and iverilog supported them at 1.0 and 0.9 separately.

xiaochuang-lxc, 2023-08-27:
> i'm kind of a bit scared adding support directly on the simulation backends XD Would kinda prefer adding it the "soft" way, which would also have the advantage to make it visible in simulation waves
> 
> I very very busy for the next 2 months, but will try to find some time to try

And Using the "soft" way may be not friendly to the code coverage

zyn810039594, 2024-02-16:
Hello,
is there any process about this issue?
I'm not familiar with VPI...but I found a solution in Cocotb. Force and Release can be supported by changing the parameter of the function `vpi_put_value`.

```C
XXTERN vpiHandle    vpi_put_value       PROTO_PARAMS((vpiHandle object,
                                                      p_vpi_value value_p,
                                                      p_vpi_time time_p,
                                                      PLI_INT32 flags));
```

The parameter `flags` can be changed to `vpiNoDelay`(which I see in the VpiPlugin.write_cmd_raw of the spinalhdl.sim lib), and `vpiForceFlag`/`vpiReleaseFlag`.

Maybe they're useful?

(I can't code with C++ so it's hard for me to solve it myself...)

Dolu1990, 2024-02-19:
Ahh interresting ^^
You probably right, it could be used aswell, but i have no idea how the VPI backend work either.
About the verilator itself, i don't know if we could use the VPI interface there aswell for that, just to keep things simple

zyn810039594, 2024-02-22:
> Ahh interresting ^^ You probably right, it could be used aswell, but i have no idea how the VPI backend work either. About the verilator itself, i don't know if we could use the VPI interface there aswell for that, just to keep things simple

I will have a try when I have time hhhhhh   maybe this weekend